carla-simulator/carla #826: example.py timesout

adarsh9975, 2018-10-02:
Platform: Windows 10
Carla : 0.9.0

The example.py times out. with following message:

[image]

I have already done following:
1) build and install python API using setup.py
 
I have already tried following:
1) Allowing ports 2000 and 2001 in firewall settings
2) Turning on/off the firewall
3) Increasing the timeout duration to upto 10 seconds

iFuSiiOnzZ, 2018-10-02:
Just to be sure, do you have Carla running, `make launch`? Or maybe the ports are used by another program, try to change the port  `-carla-port=xxx`, the executable file can be found in `Unreal\CarlaUE4\Binaries\Win64\`

adarsh9975, 2018-10-02:
changing the port worked. Thanks @iFuSiiOnzZ . Can't wait to get my hands dirty :)

closed 2018-10-02 by adarsh9975