carla-simulator/carla #828: Engine crash handling finished; re-raising signal 11 for the default handler. Good bye!!!

AliBaheri, 2018-10-02:
Hi,
Thanks for a such fantastic car simuloator.

For the first time when I want to run the simulator on Linux by `./CarlaUE4.sh /Game/Carla/Maps/Town02
`
I am facing with the follwoing message:
`Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
`

What does that mean and how can I resolve that?
Thanks.

AliBaheri, 2018-10-02:
Thanks @nsubiron 

Yes, I think the problem stems from "drivers". In fact, in additional drivers in Ubuntu, I have selected 

"Using NVIDIA binary driver- version 340.107 from NVIDIA-340 (propriety)"

But I am facing to a new message:

`Can not create Open GL context. Check that the drivers support at least Open GL 3.4 (or re-run with opengl -3)`

Should I install additional driver? If so, would you please some info?

Thanks.

nsubiron, 2018-10-02:
You can try first running `./CarlaUE4.sh -opengl3` to see if it works. Otherwise you can try installing the latest nvidia drivers for your GPU from the nvidia download page.

closed 2018-12-08 by stale[bot]

DanlanChen, 2019-03-06:
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
still get this error on linux server

HubFire, 2019-04-29:
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
:Nvidia RTX 2070+ubuntu 16.04      Nvidia Driver Version: 410.78       CUDA Version: 10.0

Stefan566, 2019-10-23:
[2019.10.23-13.17.16:706][  0]CrashReportClientLog: OnProcessRequestComplete(), State=SendingFiles bSucceeded=1
[2019.10.23-13.17.16:706][  0]CrashReportClientLog: State change from SendingFiles to SendingFiles
[2019.10.23-13.17.16:706][  0]CrashReportClientLog: All uploads done
[2019.10.23-13.17.16:706][  0]CrashReportClientLog: State change from SendingFiles to Finished
[2019.10.23-13.17.17:375][  0]LogLinux: FLinuxPlatformMisc::RequestExit(0)
[2019.10.23-13.17.17:375][  0]LogGenericPlatformMisc: FPlatformMisc::RequestExit(0)
[2019.10.23-13.17.17:408][  0]CrashReportClientLog: Final state (Receiver) = Finished
[2019.10.23-13.17.17:408][  0]CrashReportClientLog: Final state (Receiver) = Unknown UploadState value
[2019.10.23-13.17.17:408][  0]LogExit: Preparing to exit.
[2019.10.23-13.17.17:409][  0]LogExit: Object subsystem successfully closed.
[2019.10.23-13.17.17:410][  0]LogModuleManager: Shutting down and abandoning module HTTP (8)
[2019.10.23-13.17.17:443][  0]LogModuleManager: Shutting down and abandoning module SSL (7)
[2019.10.23-13.17.17:443][  0]LogModuleManager: Shutting down and abandoning module CrashDebugHelper (4)
[2019.10.23-13.17.17:443][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2019.10.23-13.17.17:443][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

Kin-Zhang, 2022-01-17:
Engine crash handling finished; re-raising signal 6for the default handler. Good bye.
I aslo met this error on carla 0.9.10.1 but it happen when I run the carla for a long time like 8 hours until have this error.

xinshuoweng, 2022-04-01:
@Kin-Zhang, Same here. Sometimes it is fine to run the process for even more than one day, but sometimes it reports this error just after one hour, kind of random and unstable. Any fix you are aware of?

Kin-Zhang, 2022-04-02:
> @Kin-Zhang, Same here. Sometimes it is fine to run the process for even more than one day, but sometimes it reports this error just after one hour, kind of random and unstable. Any fix you are aware of?

No, I didn't find any fix to this issue...

xinshuoweng, 2022-04-03:
Any help, @nsubiron?

AOOOOOA, 2022-05-31:
Hi, is there any solution to this problem?

PaGu3, 2023-03-29:
Hi, Did anyone find the solution for this problem?

Sultan91, 2023-10-14:
I found that such error are not raised if you change the map

zzj403, 2023-10-15:
Sitll meet this proble in carla 0.9.14 and almost stop at 2h~3h. During my program, I constantly destory and spawn new cars and sensors.

cihan063, 2024-02-17:
I lost my internet connection when I was working on a remote computer, and then I tried to run it, I came across this issue. Apparently, it wasn't terminated correctly when I lost my connection. 

If you are on Windows, you can try to find and terminate the process from the task manager. 

I was on Ubuntu. I wrote ''htop'' to terminal to see the processes, found the carla process, and pressed f9 to terminate and send the SIGTERM (15th option) signal to terminate. I solved it this way. I hope it works for you as well.