carla-simulator/carla #832: What is Type of Carla unreal plungin?

JSaffari, 2018-10-03:
Hi everyone
First of all I want to tanks all Carla Devs to do this project and release that for people in Github.
I am going to create a plugin like Caral beside that for some reason.
My question : What Pick-Up you for Carla plugin (meaning plugin type in unreal engine)?

nsubiron, 2018-10-03:
Hi @JSaffari,

The plugin is not a type, but a folder with a bunch of code. The plugin is defined in the `*.uplugin` file, and should be stored in the project's _Plugins_ folder. Epic has good tutorials for creating plugins, take a look at them, e.g. https://docs.unrealengine.com/en-us/Programming/Plugins

JSaffari, 2018-10-03:
Hi Dear @nsubiron 
Thanks for your replay.
I know that plugin struct in ue4 but, only want to prepare a plugin beside Carla and Add new feature for your simulation in other plugin the reason is isolation our code from your code "Carla" .

Have you any option for me?
I think that was batter way to isolation form Bad Bugs.
One another think, We will use Carla in Our simulation  in new version of simulator to learn Bus Driver and give to their a  certificate in Iran.
Would you let us to do that?
Again thank for replay.

nsubiron, 2018-10-04:
Hi @JSaffari,

It shouldn't be any problem creating another folder with a different plugin.

And of course, you can use Carla for that. Carla is distributed under MIT license which is quite open.

closed 2018-11-05 by nsubiron