carla-simulator/carla #835: LIDAR data acquisition

michellevalente, 2018-10-03:
Hi,

I'm creating a dataset using only LIDAR and position information. However, I'm having a problem in obtaining the point cloud complete (and not during a rotation). Is there any way of getting only the last complete rotation of the sensor? 

Any help would be appreciated, thanks!

nsubiron, 2018-10-03:
Hi @michellevalente,

If you can run the simulation at [fixed time-step](https://carla.readthedocs.io/en/latest/configuring_the_simulation/), you can adjust the frame rate and the Lidar frequency to exactly match one complete rotation on each frame. This way the message will contain always a complete rotation. This way also the simulation may run faster if your hardware allows it.

Though, note that if you do a complete rotation in a single frame everything will correspond to a "still image" of that frame. If you want more realism you can adjust it to match say 1/5 of the rotation, then merge 5 files into one. With the fixed frame rate you can even do crazy things like 1000FPS to capture a slow motion of the simulation.

Romanenko-Serhii, 2018-10-04:
Yes, I do like say @nsubiron for emulate true lidar spin effects and use 2000fps (2000fps have more LIDAR bims then 1000fps). Then in function _on_loop I collect every LIDAR bim, and every 2000 fps save it to file.
And, if you want a ground trues in some period of time, you can add second LIDAR sensor and collect all points in one moment (it looks like flash LIDAR) data.
You can specified PointsPerSecond and  RotationFrequency

michellevalente, 2018-10-04:
Thank you for the suggestions @nsubiron and @Romanenko-Serhii !

I'm trying to generate a dataset to pre-train a network that will use the KITTI dataset after to find results. 
I did what was recommended by @nsubiron and I was able to get the full rotation, but I'm still having a problem that maybe you guys could help me understand. Between two scans I find almost no difference, while there is significant drift between two scans from the real dataset (which is basically what I need my learning algorithm to learn). I'm taking the data at the same frequency of the real dataset, 10hz.

Do you guys have any idea why this is happening? Do you think it can be just due to the lack of the true lidar spin effect?

Romanenko-Serhii, 2018-10-04:
@michellevalente 
You need create a LIDAR with parm, like:
```
 lidar = sensor.Lidar('Lidar_one_shot')
        lidar.set_position(0, 0, 2.5)
        lidar.set_rotation(0, 0, 0)
        lidar.set(
            Channels=64,
            Range=50,
            PointsPerSecond=100000,
            RotationFrequency=1,
            UpperFovLimit=10,
            LowerFovLimit=-30)
        settings.add_sensor(lidar)
```

Then you need set FPS > 1500, like: ./CarlaUE4.sh  -windowed -ResX=800 -ResY=600 -carla-server -carla-world-port=2000 **-fps=1600**  -benchmark

And upgrade function _on_loop, for collecting all bins* into one file.

This affects will be grateful seen on the EGO that moves on oncoming traffic.

*At bins I mean one laser scan with 64 channels.

Romanenko-Serhii, 2018-10-04:
Btw, by default, your car model looks like two cubes. If you wanna get a more realistic view of the car, you need to make some changes with UnrealEngine Carla project.

michellevalente, 2018-10-04:
Why I would need FPS > 1500? In my computer is impossible to run the simulator at this configuration.

Romanenko-Serhii, 2018-10-04:
@michellevalente I see on e-mail notification about your question about FPS, but don't see it here. Do you steel need the answer?

michellevalente, 2018-10-04:
@Romanenko-Serhii 
I got it! thank you :)

closed 2018-10-04 by michellevalente

maxjaritz, 2018-11-21:
> Btw, by default, your car model looks like two cubes. If you wanna get a more realistic view of the car, you need to make some changes with UnrealEngine Carla project.

Could you explain what changes have to be made? Thank you!

Romanenko-Serhii, 2018-11-21:
@maxjaritz Content -> Blueprints -> Vehicles chose some car (Audi). There will be phisical model. When you open it in right pannel find drop-down list where current state Box, Change it to some anuother state and click on green button in right corner.

maxjaritz, 2018-11-21:
Okay, thanks!

ravishk1, 2019-04-16:
Hi,

I want to use Lidar as different distance sensor, Suppose I want to use Lidar as a Radar sensor, for that I want to restrict 360 degree rotation of the Lidar. I want it to move only in the range of suppose 0 to 180 degree. Can we modify the Lidar configuration in such a way?? Just like we restrict the Upper FOV and Lower FOV, can we restrict the Horizontal FOV??

Any help will be highly appreciated.
Thanks !

nsubiron, 2019-04-16:
@ravishk1 

> can we restrict the Horizontal FOV?

That would be a nice feature, but not currently implemented. Maybe you can try implementing it yourself, take a look at [RayCastLidar.cpp](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/RayCastLidar.cpp), this seems like a good first time contribution :wink:

vishal-prgmr, 2019-06-05:
@nsubiron , Is there no other way that a complete 360deg sweep from the lidar can be captured other than  adjusting the fps of the server?

visakhchikku, 2023-10-05:
Is it possible to save lidar data in .pcap in carla?

Petros626, 2025-06-02:
> Hi，do you know the labeling-tool for KITTI data set？

off topic, but this tool is for kitti360: https://github.com/autonomousvision/kitti360LabelTool