carla-simulator/carla #836: Error while executing Docker for Carla

Deepak3994, 2018-10-04:
Hi,

I have done all the per-requisities reduired for running the docker for carla. When i enter the docker-run command, it gets stuck. How do i proceed with that?

The docker run command is "docker run --runtime=nvidia --rm -p 2000-2002:2000-2002 -e NVIDIA_VISIBLE_DEVICES=0 carlasim/carla:0.8.2" and also have tried executing "docker run -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=0 carlasim/carla:0.8.4 /bin/bash CarlaUE4.sh"

harlowja, 2018-10-16:
Can you let me know what 'gets' stuck means? Any logs?

gy20073, 2018-11-11:
I had the same problem. it runs within the docker but the command line output does not update smoothly. It usually updates in a batch and then stop for a while and then updates in a batch again. 

from outside the docker, the carla client can not connects to it stably.

Deepak3994, 2019-01-04:
I have installed the nvidi-docker and as well as nvidia-driver on my AWS instances. While running the below "docker run -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=0 carlasim/carla:0.8.4 /bin/bash CarlaUE4.sh -carla-server" command i am getting the error as "CarlaUE4.sh: line 5: /home/carla/CarlaUE4/Binaries/Linux/CarlaUE4: Text file busy".

Kindly help.

closed 2019-03-12 by stale[bot]