carla-simulator/carla #840: How to make lidar raycasting faster

praveenputhettu, 2018-10-05:
CARLA version number : 0.8.4
Platform/OS : Ubuntu 16.04

I have mounted 5 lidars on the player vehicle, which generates 2500K points. This caused the simulation to slow down. I am using the per built binaries of CARLA version 0.8.4 on ubuntu 16.04. I am using an Nvidia 1050 Ti GPU with 32 GB of RAM. 
* Is prebuilt binaries using GPU optimization ? I found that GPU utilization is below 25% during simulation. * Do I need to built CARLA source locally in my system with GPU optimization enabled ?. OR is this a known issue, which is to be solved in future versions ?

Romanenko-Serhii, 2018-10-08:
As I understood, in feature release 0.9.1 LIDAR will be work faster. But now it did not possible.
But I may wrong.

praveenputhettu, 2018-10-16:
Thank you Romanenko-Serhii

closed 2018-12-22 by stale[bot]