carla-simulator/carla #842: Hi,

mkreddy477, 2018-10-05:


marcgpuig, 2018-10-05:
Why you want to do that? JPEG aims lossy compression, and losing some bits of your semantic segmentation or depth image can be really _bad_ for your experiments. My recommendation here is to try to read PNG.

If you want to save it in JPG/JPEG anyway, you can modify or create a new client and use a python library that allows you to save images in JPEG (like [Pillow](https://python-pillow.org/)).

Also, please fix the title of this issue, put a simple and descriptive title.

mkreddy477, 2018-10-08:
Hi @marcgpuig 

Thanks for the quick reply. Here I am using the client in a different environment where the support for reading PNG files doesn't exist.

marcgpuig, 2018-10-09:
@mkreddy477 
Cool, use Pillow to save the images in JPEG, but remember about the compression issue :)
Good luck!

closed 2018-12-10 by marcgpuig