carla-simulator/carla #845: Adding a Bus

Nozomeister, 2018-10-08:
Dear Carla team,
I tried to follow the adding new assets guide to add a new vehicle, specifically a bus.
When I tried to use several 3d models on it, when I imported the model with the skeletal mesh from General4WheeledVehicle, the output vehicle will have a buggy animation (jumping around, flying toward the sky when spawned in the map).
Can you assist me with this? I have attached a model for reference.

Thank you 
[Bus+Metro+Mini02.FBX.tar.gz](https://github.com/carla-simulator/carla/files/2454515/Bus%2BMetro%2BMini02.FBX.tar.gz)

TheNihilisticRobot, 2018-10-08:
Hi @Nozomeister

I'm sorry but i've opened the mesh you've sent to us and once I checked it seems like the skeleton you have bonded to the model is not the one we use in the project (General4WheeledVehicle) You need to extract the skeleton from the project and then bind this skeleton to the mesh in a modelling software, then import it into Unreal and proceed with the rest of the vehicle guide.

Please, come back if you keep having problems after trying this.

Nozomeister, 2018-10-08:
> Hi @Nozomeister
>
> I'm sorry but i've opened the mesh you've sent to us and once I checked it seems like the skeleton you have bonded to the model is not the one we use in the project [...]

Thank you for the answer. Probably I have misunderstood the process, and I will try to resolve it first.

Nozomeister, 2018-10-10:
@TheNihilisticRobot 
I have tried to export the General4WheeledVehicle, but it only has the option of .T3D or .COPY, which are Unreal Object Text files, which cannot be read by Blender. Is there any way to export this skeletal mesh as FBX file?

TheNihilisticRobot, 2018-10-10:
Yes, there is. You need to find any of our vehicles inside the content browser rightclick-it and select AssetActions-Export. This will give you the vehicle as an .FBX. Open or import it in the modelling software you have, delete the geometry and you'll end up with the skeleton you need.

Nozomeister, 2018-10-12:
@TheNihilisticRobot Thank you for the assist. I have tried to do that in the 3d software and tried to import back the 3d model FBX, but the skeletal mesh failed to merge and the log said all of my vertices had no weight against them, but I have done the weight paint as well as vertex paint.
I have attached the FBX file for reference. Thank you so much.
[FBX.tar.gz](https://github.com/carla-simulator/carla/files/2471349/FBX.tar.gz)

closed 2018-12-18 by stale[bot]

YDSh510, 2019-02-27:
@Nozomeister - Hi,

Are you able to export 4-wheeled skeleton file as .fbx . There is no such option in content browser rightclick-it and select AssetActions-Export. 

It only gives .T3D or .COPY option.

Can you please share if you have the same. Thanks