carla-simulator/carla #847: Carla + Apollo [ros_bridge]

saurabhjha1, 2018-10-08:
When configuring Carla and Apollo together, I get the following warning messages and no display on dreamview. Can someone help me fix it?

I am using Carla 0.8.4 and Apollo 3
``
[WARN] [1539009430.077119, 1.283000]: Could not process inbound connection: topic types do not match: [std_msgs/String] vs. [visualization_msgs/Marker]{'message_definition': 'string data\n', 'callerid': '/carla_to_apollo_data', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/player_vehicle', 'node_type': 'rospy', 'type': 'std_msgs/String'}
[WARN] [1539009431.078609, 2.291000]: Could not process inbound connection: topic types do not match: [std_msgs/String] vs. [visualization_msgs/Marker]{'message_definition': 'string data\n', 'callerid': '/carla_to_apollo_data', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/player_vehicle', 'node_type': 'rospy', 'type': 'std_msgs/String'}
[WARN] [1539009433.083043, 4.250000]: Could not process inbound connection: topic types do not match: [std_msgs/String] vs. [visualization_msgs/Marker]{'message_definition': 'string data\n', 'callerid': '/carla_to_apollo_data', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/player_vehicle', 'node_type': 'rospy', 'type': 'std_msgs/String'}
[WARN] [1539009437.084500, 8.266000]: Could not process inbound connection: topic types do not match: [std_msgs/String] vs. [visualization_msgs/Marker]{'message_definition': 'string data\n', 'callerid': '/carla_to_apollo_data', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/player_vehicle', 'node_type': 'rospy', 'type': 'std_msgs/String'}
[WARN] [1539009445.093683, 16.298000]: Could not process inbound connection: topic types do not match: [std_msgs/String] vs. [visualization_msgs/Marker]{'message_definition': 'string data\n', 'callerid': '/carla_to_apollo_data', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/player_vehicle', 'node_type': 'rospy', 'type': 'std_msgs/String'}
[WARN] [1539009461.111044, 32.330000]: Could not process inbound connection: topic types do not match: [std_msgs/String] vs. [visualization_msgs/Marker]{'message_definition': 'string data\n', 'callerid': '/carla_to_apollo_data', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/player_vehicle', 'node_type': 'rospy', 'type': 'std_msgs/String'}
[WARN] [1539009493.144355, 64.300000]: Could not process inbound connection: topic types do not match: [std_msgs/String] vs. [visualization_msgs/Marker]{'message_definition': 'string data\n', 'callerid': '/carla_to_apollo_data', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/player_vehicle', 'node_type': 'rospy', 'type': 'std_msgs/String'}
[WARN] [1539009557.206657, 128.397000]: Could not process inbound connection: topic types do not match: [std_msgs/String] vs. [visualization_msgs/Marker]{'message_definition': 'string data\n', 'callerid': '/carla_to_apollo_data', 'tcp_nodelay': '0', 'md5sum': '992ce8a1687cec8c8bd883ec73ca41d1', 'topic': '/player_vehicle', 'node_type': 'rospy', 'type': 'std_msgs/String'}
``

luqiang21, 2018-10-12:
> When configuring Carla and Apollo together, I get the following warning messages and no display on dreamview. Can someone help me fix it?
>
> I am using Carla 0.8.4 and Apollo 3
>
> ```
> [WARN] [...]

Hi Saurabh, how do you managed to connect Carla and Apollo? Can you tell me roughly the steps? Thank you !

closed 2018-12-18 by stale[bot]

yangydavid, 2019-01-10:
@saurabhjha1  Hi, did you manage to solve the problem? I have some questions about how to connect Carla with Apollo through rosbridge, would you mind sharing some of your thoughts? Thanks!

volprjir, 2019-04-03:
Hello, has anyone succeed with connecting Carla to Apollo? I followed this manual: https://nemodrive-docs.readthedocs.io/projects/nemodrive-apollo/en/latest/config_run.html

But it seems to me there are no details and in addition this does not work with Carla 0.9.X. My final point I got is I can access Dreamview where Carla option is between Map selection. But the car does not show up in that portal. Please if anyone knows how to do it, let me know

daohu527, 2019-04-11:
I have the same problem.
Currently, Apollo uses the cyber framework to send messages. I will review the changes and see if the original ros-bridge is still available.

volprjir, 2019-04-11:
@daohu527 Ok, but when I use the apollo from nemodrive repo, it should still use ROS, is it right?

daohu527, 2019-04-11:
> @daohu527 Ok, but when I use the apollo from nemodrive repo, it should still use ROS, is it right?

Yes, the nemodrive repo is use Apollo 3.0, which still use ROS for connect. The Apollo 3.5 and above will use cyber instead.

volprjir, 2019-04-11:
@daohu527 ok, so it means that the bridge provided from nemodrive should work. But I'm not sure why it does not work. Is there any difference in messages provided by Carla and the ones which Apollo requires?