carla-simulator/carla #848: How to run example with compiled version on linux with python 3

versatran01, 2018-10-09:
Hi I was wondering how one could run the example.py with the compiled version on linux.
The only thing I found in the downloaded package is an egg file for python 2.7. Does this mean I have to build from source to run carla in python3?

marcgpuig, 2018-10-09:
> Does this mean I have to build from source to run Carla in python3?

Yes it does :)
But you only must compile LibCarla and the PythonAPI, so:

    git clone https://github.com/carla-simulator/carla.git
    cd carla
    make PythonAPI

should generate the `.egg` file that you need.

Cheers!

versatran01, 2018-10-09:
Thanks for the help, but the build failed due to some boost issue
```
../../LibCarla/cmake/../source/carla/Buffer.h:204:42: error: no member named 'is_const_buffer_sequence' in namespace 'boost::asio'
    typename std::enable_if<boost::asio::is_const_buffer_sequence<T>::value>::type
```
Compiler is clang5.0, cmake version 3.12.3 and ubuntu 16.04

Manojbhat09, 2018-10-16:
@versatran01  i am facing the same issue, were you able to fix it?

versatran01, 2018-10-16:
@Manojbhat09 No I didn't,

Manojbhat09, 2018-10-16:
I am using windows WSL to run it, do you think thats the issue?

versatran01, 2018-10-17:
This seems to be a boost or compiler issue.

marcgpuig, 2018-10-19:
Sorry I was really busy!
@Manojbhat09 thanks for your feedback :)
@versatran01 did you solve it?

JoveIC, 2018-10-24:
@marcgpuig  Hi, I followed your instruction, and it showed "BuildPythonAPI.sh: Success!"
I think I was able to build the .egg file, I found it located under carla/PythonAPI/dist, but I still only see a "carla-0.9.0-py2.7-linux-x86_64.egg" file there.

Am I in the right directory? Any thoughts on this?

marcgpuig, 2018-10-24:
@versatran01 @JoveIC
I'm sorry but I got confused! `master` does not support python 3 currently.
0.9.1 will do!