carla-simulator/carla #850: Got "ERROR: tcpserver 0 : connection failed: Operation canceled" running a compiled-from-source-code carla 0.8.2 or 0.8.4

sunpochin, 2018-10-10:
Got "ERROR: tcpserver 0 : connection failed: Operation canceled" running a compiled-from-source-code carla 0.8.2 or 0.8.4

However if I run the pre-compiled release version everything seems ok.
Any thought is appreciated, thanks!

./CarlaUE4.sh -carla-settings=./CarlaSettings.ini -carla-server -windowed -ResX=1600 -ResY=1200
4.18.3-0+++UE4+Release-4.18 514 0
Disabling core dumps.
ERROR: tcpserver 0 : connection failed: Operation canceled 
ERROR: tcpserver 0 : connection failed: Operation canceled 
ERROR: tcpserver 0 : error reading message: Bad file descriptor

MaxTD, 2018-11-22:
Hey Guys I'm trying to setup 0.8.2 source code for UE4 4.19
But when i try to run Rebuild.bat nothing shows up

sunpochin, 2018-11-29:
For anyone has similar issues:
I did following things:
1. un-installed clang-5.0, make sure I am using clang-3.9 only.
2. Followed https://carla.readthedocs.io/en/stable/how_to_build_on_linux/
Remove UnrealEngine_4.18 folder and git clone it again, re-compiled it again.

3. Remove carla-0.8.4 branch folder and git clone it again, re-compiled it again.
After these steps my CARLA-0.8.4-dirty after packaging doesn't show "ERROR: tcpserver 0" anymore! :D

closed 2019-02-04 by stale[bot]