carla-simulator/carla #851: Problem with semantic segmentation in carla 0.9.0

ThierryDeruyttere, 2018-10-10:
Hi,

I'm trying to get the semantic segmentations from the carla simulator by using the Python API. 
I'm using the following python code to create two camera's: one that captures the current RGB view and the other one should get the semantic segmentation of that view.

```python
...

def save_to_disk(image, name):
    """Save this image to disk (requires PIL installed)."""

    filename = '{}/{:0>6d}_{:s}.png'.format(name, image.frame_number, image.type)

    image = PImage.frombytes(
        mode="RGBA",
        size=(image.width, image.height),
        data=image.raw_data,
        decoder_name='raw')

    color = image.split()
    image = PImage.merge("RGB", color[2::-1])

    folder = os.path.dirname(filename)
    if not os.path.isdir(folder):
        os.makedirs(folder)
    image.save(filename)

def save_semantic_img(image, name):
    res = labels_to_cityscapes_palette(image)

    filename = '{}/{:0>6d}_{:s}.png'.format(name, image.frame_number, image.type)

    image = PImage.fromarray(res)

    folder = os.path.dirname(filename)
    if not os.path.isdir(folder):
        os.makedirs(folder)
    image.save(filename)

bp = world.get_blueprint_library()
blueprint = bp.find('sensor.camera')
blueprint.set_attribute('post_processing', 'SceneFinal')
LOC = Location(x=155.5, y=55.8, z=39)
transform = Transform(LOC, Rotation(yaw=90))
cam1 = world.spawn_actor(blueprint, transform)
cam1.listen(lambda image: save_to_disk(image, "scene"))

blueprint2 = bp.find('sensor.camera')
blueprint.set_attribute('post_processing', 'SemanticSegmentation')
transform = Transform(LOC, Rotation(yaw=0))
cam2 = world.spawn_actor(blueprint, transform)
cam2.listen(lambda image: save_semantic_img(image, "semantic"))
...
```

These are the images I get:
[image]

[image]

As you can see the semantic image looks really weird but I have no idea how this came to be. Did anyone have the same problem?

nsubiron, 2018-10-11:
Interesting, I would expect the old image converter to work with the new API, it seems it's mixing columns and rows.

In any case, I recently ported these methods to the new API (PR #855), if you can use that branch that works pretty well and fast. It will be included in the next release, probably ready by the end of next week.

ThierryDeruyttere, 2018-10-16:
So I've tried the new API out (merged pr #815 and pr #821). I try to get the semantic segmentation by using the following code.
```python
...
i = 1
def save_semantic_img(image, name):
    global i
    image.convert(ColorConverter.CityScapesPalette)
    image.save_to_disk(name+"/{}.png".format(i))
    i+=1

blueprint2 = bp.find('sensor.camera.semantic_segmentation')
#blueprint.set_attribute('post_processing', 'SemanticSegmentation')
transform = Transform(LOC, Rotation(yaw=90))
cam2 = world.spawn_actor(blueprint, transform)
cam2.listen(lambda image: save_semantic_img(image, "semantic"))
```
gives me the following image:
[image]

nsubiron, 2018-10-17:
Hi @ThierryDeruyttere,

That looks familiar, it's a depth image converted to CityScapes palette. I'm guessing your `blueprint` variable is a depth camera :wink: 
```py
blueprint2 = bp.find('sensor.camera.semantic_segmentation')
^~~~~~~~~~
cam2 = world.spawn_actor(blueprint, transform)
                         ^~~~~~~~~
```

And btw, if you convert the image when saving is slightly more efficient as it doesn't have to modify the image
```py
image.save_to_disk('filename', ColorConverter.CityScapesPalette)
```

ThierryDeruyttere, 2018-10-17:
Hi @nsubiron,

Oops that is quite embarrassing  😄. 
I changed it to blueprint2 but now I just get a black image... (blueprint was a rgb sensor btw :p) 
Also when i'm using the rgb sensor and I just save it (so no color conversion), I get a white image.

Also maybe a question about the API, will the different sensors remain as a blueprint in python? (thus sensor.camera.rgb, sensor.camera.depth, ...) Because I don't see why you would need to have the different blueprints and still have the option do to color conversions?

nsubiron, 2018-10-17:
Oh true, it's not depth is rgb :D

Are you talking only about the first frame? cause first one usually comes empty. Otherwise is strange, it works on all the tests I've made.

Having them as different sensors IMO is clearer as people usually struggled to find the post-processing attribute, plus internally it's easier to implement. But all of the cameras produce the same data, an "Image" object, then this images can be converted to anything, or use them as they come. In some cases, depth for instance, you have two conversions to choose from; and possibly in the future we have a different sem-seg palette for instance.

ThierryDeruyttere, 2018-10-18:
I did more tests and I found out that when using rgb camera with a color conversion gives me some output which you can see below.

Depth:
[image]

LogarithmicDepth:
[image]

but if I use None, it remains white.

nsubiron, 2018-10-18:
Hi @ThierryDeruyttere,

Hmm it looks like it could be the alpha of the image, we had issues with this in the past, but the approach we have now seemed to work fine on Linux and Windows, maybe not so much on Mac.

Could you please try applying these changes and recompiling PythonAPI? Basically I'm trying to force the saving methods to ignore the alpha channel.

```diff
diff --git a/LibCarla/source/carla/image/ImageView.h b/LibCarla/source/carla/image/ImageView.h
index ef1a57c..f04839e 100644
--- a/LibCarla/source/carla/image/ImageView.h
+++ b/LibCarla/source/carla/image/ImageView.h
@@ -48,11 +48,13 @@ namespace image {
     }
 
     static auto MakeView(sensor::data::ImageTmpl<sensor::data::Color> &image) {
-      return MakeViewFromSensorImage<boost::gil::bgra8_pixel_t>(image);
+      return boost::gil::color_converted_view<boost::gil::bgr8_pixel_t>(
+          MakeViewFromSensorImage<boost::gil::bgra8_pixel_t>(image));
     }
 
     static auto MakeView(const sensor::data::ImageTmpl<sensor::data::Color> &image) {
-      return MakeViewFromSensorImage<boost::gil::bgra8c_pixel_t>(image);
+      return boost::gil::color_converted_view<boost::gil::bgr8c_pixel_t>(
+          MakeViewFromSensorImage<boost::gil::bgra8c_pixel_t>(image));
     }
 
     template <typename SrcViewT, typename DstPixelT, typename CC>
```

ThierryDeruyttere, 2018-10-18:
Hi @nsubiron,

I'm afraid this didn't work, all images are black now instead of white.

ThierryDeruyttere, 2018-10-18:
Hi @nsubiron,

Thanks to your suggestion I got it to work (in a bit of a hack though).
I removed the change you proposed and just removed the alpha channel in python.
```python
def save_to_disk(image, name):
    global j
    """Save this image to disk (requires PIL installed)."""
    #image.convert(ColorConverter.CityScapesPalette)
    #print image.raw_data
    array = np.frombuffer(image.raw_data, dtype=np.dtype("uint8"))
    array = np.reshape(array, (image.height, image.width, 4))
    array = array[:, :, :3]
    array = array[:, :, ::-1]
    scipy.misc.imsave(name+'/{}_rgb.png'.format(j), array)
    j+=1
```
This gives me the correct image.
[image]

But you are right, apparently the alpha channel creates a problem on mac.

nsubiron, 2018-10-18:
Ok, well good to know that it's the alpha. I'll open an issue referencing this so we can investigate a solution that works on every platform.

ThierryDeruyttere, 2018-10-18:
@nsubiron Sounds good! I think we can close this for now then?

nsubiron, 2018-10-18:
Ok, perfect

closed 2018-10-18 by ThierryDeruyttere

Deepak3994, 2018-12-19:
> Hi @nsubiron,
> 
> I just did some more tests. I can confirm that the segmentation camera works fine (check below). So thanks a lot for that! :D
> [image]
> It only does weird things when I spawn static meshes through python (custom code). The semantic image remains black then. So I'll need to check my code out. On the other hand for the rgb sensor all images remain white (tested this on Town1 at x=155.5, y=55.8, z=39) not only the first one. It also stays white when spawning static meshes but this is more of a problem on my end.
> 
> > Having them as different sensors IMO is clearer as people usually struggled to find the post-processing attribute, plus internally it's easier to implement. But all of the cameras produce the same data, an "Image" object, then this images can be converted to anything, or use them as they come. In some cases, depth for instance, you have two conversions to choose from; and possibly in the future we have a different sem-seg palette for instance.
> 
> Okay fair enough :)

Hi @ThierryDeruyttere 
Could you share the code or steps how to extract the semantic segmentation images from carla simulator.

ThierryDeruyttere, 2018-12-19:
@Deepak3994 
Hi there,

You can find the code that I was using at: https://github.com/ThierryDeruyttere/carla-simulator-mac/blob/master/mesh.py

zhangjunwang, 2019-11-03:
I have the same problem! 
I use Carla 0.9.5 in linux,. I change the code in PythonAPI.examples.manual_control_steeringwheel.py to save semantic_segmentation image.
and I got this image:
[image]

The normal scene is:
[image]

I just use this code to save it:
`image_semseg.convert(cc.CityScapesPalette)`
`image_semseg.save_to_disk('_out_seg/%08d' % image.frame_number)`

Could you help me to solve it?
Very thanks!

pirate-lofy, 2020-01-25:
I have another issue with the semantic segmentation sensor... I'm just taking the image from the sensor and send it to the labels_to_cityscapes_palette() function and getting this..
[image]

PR #855 (2018-10-23): Add IO methods for sensor data
#### Description

Requires #815.

Adds the following methods to the API:

  * Image: `save_to_disk(filename, color_converter=None)`, saves the image to disk applying the given color converter. Available formats, PNG, TIFF and JPEG (JPEG has losses though).
  * Image: `convert(color_converter)` Apply the given color converter in place.
  * LidarMeasurement: `save_to_disk(filename)` Saves the point cloud in PLY format.

The following color converters are available:

- `None`
- `Depth`
- `LogarithmicDepth`
- `CityScapesPalette`

Related issues:

- Fix #821.
- Fix #7 [... cut]