carla-simulator/carla #852: Changing the weather from PythonClient

mhusseinsh, 2018-10-10:
Hello
Is there a way to change the weather from the PythonClient instead of using the keyboard button 'C'

So is there a function that I can call to do so ?

iFuSiiOnzZ, 2018-10-10:
Currently there is no way to change the weather from the client, in the future it will be available.

mhusseinsh, 2018-10-11:
Hello @iFuSiiOnzZ 
Thanks for your reply .. well that’s bad 

Okay I will tell you what I want, maybe you have an idea to help

Simply, I want to extract paired dataset from CARLA with two different weather conditions
So that’s why i was thinking to flip the weather everytime during the same frame, so i can save exactly the same scene two times with different weathers

The problem, that for each episode, the car starts at a random position in the map, as well as other non-player agents, so there is no guarantee if I run two different scenarios with different weathers, I will have a paired image

Do you have any idea or hint on how to do so ?

iFuSiiOnzZ, 2018-10-11:
What you can do is to generate a `CarlaSetting.ini` with a fixed seed for cars (`SeedVehicles`) and pedestrians (`SeedPedestrians`). Once that you have the settings file, reset the episode changing the weather in the settings file.

Take a look at [CARLA Settings](https://carla.readthedocs.io/en/stable/carla_settings/) documentation, and use the example file modified at your need or generate one using the `settings.py` from `PythonClient/carla/`.

PS: Take also a look at the `PythonClient/client_example.py` and play a bit with the settings.

mhusseinsh, 2018-10-11:
@iFuSiiOnzZ 
Thanks for your reply

I did a workaround already
From the `manual_control.py`, for every new episode I change the weather from 1 to 8, and I did removed the `randomize_seed` and `randomize_weather` as well as randomizing the player start position every 2 episodes, so now (techincally) I have 2 consecutive exact episodes, one in daytime, and one in sunset, and the player start position is randomized every 2 episodes

Why I asked about changing the weather for every frame not episode, is that I am running in autopilot mode, so if I want to have the exact same episodes, this won't work

I am not sure, how the car will react as actions (speed, turning right/left, braking) and so between two episodes

Is there a work around on this issue ?

iFuSiiOnzZ, 2018-10-11:
If you have a fixed seed and start the car at the same point you will have exactly the same, as for now the weather is just a visual effect, doesn't affect the car physics.

mhusseinsh, 2018-10-11:
@iFuSiiOnzZ So even the actions of the car will be the same in the autopilot mode ? I will have the same episode exactly in the two weather modes ?

nsubiron, 2018-10-11:
Hi @mhusseinsh,

Let me comment a bit more about repeatability between episodes.

First of all, you need to make sure:
 1. You run the simulator at [fixed frame-rate](https://carla.readthedocs.io/en/latest/configuring_the_simulation/).
 2. You set the random seeds both for vehicles and pedestrians.

With that you should be able to get the exact images pixel-by-pixel for about 10 min at least, it depends on the number of pedestrians you add. That's because the pedestrians follow Unreal's navigation system and we don't have full control over that, so they're not fully deterministic. Eventually some pedestrian will diverge and you see these small differences. If you remove pedestrians it should be fully repeatable.

Check out [this video](https://youtu.be/xU80o0Bdh-U) we made a while ago, it is a pixel-by-pixel comparison of two runs starting after about 15 min of simulation. There you can see how pedestrians diverge and propagate it to vehicles.

mhusseinsh, 2018-10-11:
@nsubiron 
Thanks for your reply

> First of all, you need to make sure:
> 
> 1. You run the simulator at [fixed frame-rate](https://carla.readthedocs.io/en/latest/configuring_the_simulation/).

Can this be set and defined from the python client ? or I have to pass these flags `-benchmark -fps=5` when I start the server itself 
> 2. You set the random seeds both for vehicles and pedestrians.
> 
So I did like that now:
```py
if start == True:
            #print('random seeds')
            player_start = np.random.randint(number_of_player_starts)
            start_position = player_start
            print('Starting new episode...')
            print('Episode: ' + str(episode) + ' DAY')
            bikes = []
            self.client.start_episode(player_start)
        else:
            if weather == 1:
                player_start = np.random.randint(number_of_player_starts)
                start_position = player_start
                seed = random.getrandbits(16)
                self._carla_settings.set(SeedPedestrians=seed)
                self._carla_settings.set(SeedVehicles=seed)
                #print('random seeds')
                print('Starting new episode...')
                print('Episode: ' + str(episode) + ' DAY')
                bikes = []
                self.client.start_episode(player_start)
            else:
                #print('same seeds')
                print('Episode: ' + str(episode) + ' NIGHT')
                bikes = []
                self._carla_settings.set(SeedPedestrians=seed)
                self._carla_settings.set(SeedVehicles=seed)
                self.client.start_episode(start_position)
```
As I found out in the `randomize_seed()` function is `settings.py` that each chooses a different value for pedestrians and vehicles, so I defined the value by myself .. but look at these two examples

[image]
[image]

In the day image, there is a car, and in the night there is no car

And for these two
[image]
[image]
If I zoomed in to the casrs
[image]
[image]
You will notice, that the colors of the cars are changed
In the first image, it has two red cars, while in the second image, one is red and one is blue

> With that you should be able to get the exact images pixel-by-pixel for about 10 min at least, it depends on the number of pedestrians you add. That's because the pedestrians follow Unreal's [...]

So this means that even with what you said above, there will still be a divergence in the pedestrians, right ?

SpakMiguel, 2018-10-11:
I am also interested in the topic. Autopilot does make the car behave the same all the time if I run different episodes starting at the very same initial spot? My idea is to mount four cameras onto the car and run four episodes in order to capture all the generated data because otherwise it would be to computationally heavy. All episodes must be exactly the same.

nsubiron, 2018-10-11:
@mhusseinsh, at which point are you doing the `client.load_settings(carla_settings)`? You should call it each time you change the settings.

> I have to pass these flags -benchmark -fps=5 when I start the server itself

Yes, unfortunately you need to pass the  -benchmark -fps=X arguments, cannot be done from Python.

> So this means that even with what you said above, there will still be a divergence in the pedestrians, right?

Actually I've never tried with the command-line argument that @analog-cbarber suggests, I always thought it would be an issue of the navigation algorithm, worth giving it a shot.

And that issue with vehicles changing colors looks like a bug, let us take a look (@TheNihilisticRobot have you seen this?).

@SpakMiguel, each vehicle has it's own random number generator (derived from the global one), so they follow a fixed route determined by the seed.

mhusseinsh, 2018-10-11:
> @mhusseinsh, at which point are you doing the `client.load_settings(carla_settings)`? You should call it each time you change the settings.
Ok this was my mistake, but again
Look what I did
```
seed = random.getrandbits(16)
    def _on_new_episode(self):
        global start
        global weather
        global start_position
        global seed
        global episode
        #self._carla_settings.randomize_seeds()
        if start == True:
            self._carla_settings.set(WeatherId=weather)
        elif start == False:
            if (weather == 8):
                weather = 1
                seed = random.getrandbits(16)
                self._carla_settings.set(SeedPedestrians=seed)
                self._carla_settings.set(SeedVehicles=seed)
                self._carla_settings.randomize_seeds()
            else:
                weather = 8
                self._carla_settings.set(SeedPedestrians=seed)
                self._carla_settings.set(SeedVehicles=seed)
                self._carla_settings.set(WeatherId=weather)
        scene = self.client.load_settings(self._carla_settings)
        if self._display_map:
            self._city_name = scene.map_name
        number_of_player_starts = len(scene.player_start_spots)
        if start == True:
            #print('random seeds')
            player_start = np.random.randint(number_of_player_starts)
            start_position = player_start
            print('Starting new episode...')
            print('Episode: ' + str(episode) + ' DAY')
            self.client.start_episode(player_start)
        else:
            if weather == 1:
                player_start = np.random.randint(number_of_player_starts)
                start_position = player_start
                #print('random seeds')
                print('Starting new episode...')
                print('Episode: ' + str(episode) + ' DAY')
                self.client.start_episode(player_start)
            else:
                #print('same seeds')
                print('Episode: ' + str(episode) + ' NIGHT')
                self.client.start_episode(start_position)
        self._timer = Timer()
        self._is_on_reverse = False
```
So theoretically, I did what you suggested, but unfortunately, this happened
Sometimes it is correct, like this
[image]
[image]
And others not, like this
[image]
[image]

Correct me if I am doing something wrong

> Yes, unfortunately you need to pass the -benchmark -fps=X arguments, cannot be done from Python.
Okay perfect (Y) I will do so

> Actually I've never tried with the command-line argument that @analog-cbarber suggests, I always thought it would be an issue of the navigation algorithm, worth giving it a shot.
@analog-cbarber could you share with us the command line ?

> And that issue with vehicles changing colors looks like a bug, let us take a look (@TheNihilisticRobot have you seen this?).
Thanks, inform me when you have updates
> @SpakMiguel, each vehicle has it's own random number generator (derived from the global one), so they follow a fixed route determined by the seed.
and each vehicle is including the player vehicle itself ? or only non-player agents ?

analog-cbarber, 2018-10-11:
The command line arguments to Unreal are described here: https://docs.unrealengine.com/en-us/Programming/Basics/CommandLineArguments.

There is a `-FIXEDSEED=<seed>` option that might do the trick.

mhusseinsh, 2018-10-11:
@analog-cbarber thanks a lot, this can be helpful for sure
Anyways for me, I followed what @nsubiron suggested for doing a fixed time-step and in my case for only saving the first frame of every episode, there is no divergence

@nsubiron just the seed problem is still existing, the idea of why I want this, is because for every 2 episodes, I am starting at a random position, so maybe once, I start at a position which I have already used before, then if the seed is a fixed number, I will have the same frame exactly I had before, that's why I am trying to solve the issue of randomizing the seed every 2nd episode
You can check my code above and inform me about what is causing the problem

closed 2018-12-17 by stale[bot]