carla-simulator/carla #858: is there a way to directly save ground truth?

yyuuliang, 2018-10-10:
I know there is a segmentation camera and I have an image data with different class ids. Is there a function which can directly output the ground truth? E.g., a list of bounding box with coordinates and class ids.  So that I can just use this ground truth data and the original image data to train some DNN models?

Thanks.

yyuuliang, 2018-10-12:
After some research, I think there is no such function in the Carla code and I have to write my own function to do the work. Is this correct?
Sorry for the duplicate issue. I think this can be closed.

marcgpuig, 2018-10-20:
Indeed you must do them yourself. But of course, anything you do that you think it will be useful for anyone, feel free to share it with the community. People here will be grateful :)

Cheers!

closed 2018-10-20 by marcgpuig