carla-simulator/carla #862: Change Weather to Night in 0.8.4

mhusseinsh, 2018-10-11:
Hello
I am working with CARLA release 0.8.4

Is there a way to have a night weather ? Is there a place where I can add or change weather settings ?

Romanenko-Serhii, 2018-10-11:
Yes, look this https://github.com/carla-simulator/carla/issues/525

mhusseinsh, 2018-10-15:
@TheNihilisticRobot @nsubiron any help here please ?

Romanenko-Serhii, 2018-10-16:
@mhusseinsh He, sorry for the late response: https://carla.readthedocs.io/en/0.8.4/carla_settings/

mhusseinsh, 2018-10-16:
@Romanenko-Serhii No problem, thank you

Sorry I don't get your point, what does this has to do with my issue ? I want to achieve a night mode, but I am directly using the 0.8.4 release, which is already built (so I have the executable file directly), this means, that I don't have Config files and I cannot do the solution which is done here #525

Romanenko-Serhii, 2018-10-16:
@mhusseinsh Yes, sorry I forget to add, that you need use full version of the 0.8.4 release, not prebuilt.

mhusseinsh, 2018-10-16:
@Romanenko-Serhii so there is no way around at all to use the prebuilt ?

Romanenko-Serhii, 2018-10-16:
@mhusseinsh I think no because you need to make configuration of the server.

analog-cbarber, 2018-10-16:
Not sure how meaningful night mode would be without dynamic headlights, street lights and lighting in the buildings.

mhusseinsh, 2018-10-17:
@analog-cbarber I am talking about night mode as sun wise and sky color wise
No problem with street lights and lighting in buildings or cars, I want a normal night mode like what it is in real life, dark sky and no sun

like these examples
[image]
[image]
[image]

analog-cbarber, 2018-10-17:
Those examples clearly have dynamic lights and have building lighting, so I don't understand what you are asking for. If you turn off all sky lighting you will be driving around in the dark without lights. That's my point. Unless you also add support for dynamic headlights, street lights and modify the building meshes to show window lighting, you will not get anything like what you see above, and none of those will happen magically by playing with the weather settings.

mhusseinsh, 2018-10-17:
@analog-cbarber so in CARLA, there is no option to turn off sky lighting, and turn on street lights or car lights ? this never happens ?

analog-cbarber, 2018-10-17:
I don't believe there are any lights to turn on. You would have to create dynamic lights for every vehicle, and I don't believe they have done that yet, but you should go ahead and play with the weather settings and see if it gives you something acceptable.

mhusseinsh, 2018-10-17:
@analog-cbarber yes I will try anyways, thanks

@Romanenko-Serhii when I add a new weather in the weather config file, so this for example will be weatherID 16, do I need to rebuild again and create a new executable file ? or it will automatically be activated ?