carla-simulator/carla #864: CMAKE_C_COMPILER error

medusa606, 2018-10-11:
Installing carla 0.90 in win 7(x64), please help.

Have installed:
Unreal Engine 4.20 (couldn't find 4.19)
Visual Studio 2017 
Windows SDK 10 
Windows SDK 8.1
make
cmake
python 3

Make and Cmake accessible from command prompt (using x86 Native Tools). Getting error:
No CMAKE_C_COMPILER could be found
No CMAKE_CXX_COMPILER could be found

I have tried:

1) adding vs2017 to environment variable PATH (but i guess shouldn't need to in the  native Tools Command prompt?)

2) adding set command to cmakelist.txt in carla eg:

set(CMAKE_C_COMPILER "E:/Program Files E/Visual Studio 

IDE/VC/Tools/MSVC/14.15.26726/bin/Hostx64/x64")
set(CMAKE_CXX_COMPILER "E:/Program Files E/Visual Studio 

IDE/VC/Tools/MSVC/14.15.26726/bin/Hostx64/x64")

3) setting compiler using Cmake -D gave a parse error, eg:

cmake -D CMAKE_C_COMPILER="E:/Program Files E/Visual Studio IDE/VC/Tools/MSVC/14.15.26726/bin/Hostx64/x64/cl.exe" -D CMAKE_CXX_COMPILER "E:/Program Files E/Visual Studio IDE/VC/Tools/MSVC/14.15.26726/bin/Hostx64/x64/cl.exe" E:/carla/CMakeLists.txt

output:

E:\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "E:/carla/Build/"
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "E:/carla/Build/"
    -[install_rpclib]: Not cloning rpclib because already exists a folder called
 "rpclib-src".
    -[install_rpclib]: Generating build...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "E:/carla/Build/rpclib-src/build/CMakeFiles/CMakeOutput.log".
See also "E:/carla/Build/rpclib-src/build/CMakeFiles/CMakeError.log".

    -[install_rpclib]: [CMAKE ERROR] An error ocurred while executing the cmake.

    -[install_rpclib]: [CMAKE ERROR] Possible causes:
    -[install_rpclib]:                - Make sure "CMake" is installed.
    -[install_rpclib]:                - Make sure it is available on your Window
s "path".
    -[install_rpclib]:                - CMake 3.9.0 or higher is required.
    -[install_rpclib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows"
issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

E:\carla>make -version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32

E:\carla>cmake -version
cmake version 3.13.0-rc1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

E:\carla>

iFuSiiOnzZ, 2018-10-11:
First of all be sure you have in the `Windows PATH`, `make`, `cmake` and `python` which I understand you have it seeing you output. Second, the compiler has to be x64, if you type `cl` in `cmd` it sould give you something like this: `Microsoft (R) C/C++ Optimizing Compiler Version 19.15.26730 for x64` plus other info.

If you want to build from command line load the environment using the `vcvarsall.bat` as it sets all paths needed by `cl`. In my case the path for `vcvars.bat` is `"C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\VC\Auxiliary\Build\vcvars64.bat"`, but on `Start Menu` VS2017 should have an entry called `****x64 for VS 2017`.

medusa606, 2018-10-11:
So running cl gives:

E:\carla>cl
Microsoft (R) C/C++ Optimizing Compiler Version 19.15.26730 for x64
Copyright (C) Microsoft Corporation.  All rights reserved.
usage: cl [ option... ] filename... [ /link linkoption... ]

and running vcvars64.bat gives:
[vcvarsall.bat] Environment initialized for: 'x64'
but I get the same make error:

E:\carla>cd E:\Program Files E\Visual Studio IDE\VC\Auxiliary\Build

E:\Program Files E\Visual Studio IDE\VC\Auxiliary\Build>vcvars64.bat
**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.8.7
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

E:\Program Files E\Visual Studio IDE\VC\Auxiliary\Build>cd E:\carla

E:\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "E:/carla/Build/"
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "E:/carla/Build/"
    -[install_rpclib]: Not cloning rpclib because already exists a folder called
 "rpclib-src".
    -[install_rpclib]: Generating build...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!

iFuSiiOnzZ, 2018-10-11:
Everything seems to be in order, do you have the `C++/CLI` installed? Take a look at [this](https://gist.github.com/andschwa/391a82050118ea2862858a8a34e115c5), could be related.

medusa606, 2018-10-12:
As stated by greazer in https://gist.github.com/andschwa/391a82050118ea2862858a8a34e115c5 VS is deferring to inbuilt CMake director version 3.11

add the VS2017 CMake location to PATH:
E:\Program Files E\Visual Studio IDE\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin

but problem still persists:

**********************************************************************
** Visual Studio 2017 Developer Command Prompt v15.8.7
** Copyright (c) 2017 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

E:\Program Files E\Visual Studio IDE\VC\Auxiliary\Build>cd E:\carla

E:\carla>make launch
-[Setup]: [Batch params]: --boost-toolset msvc-14.1
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.1
-[Setup]: Install directory:  "E:/carla/Build/"
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "E:/carla/Build/"
    -[install_rpclib]: Not cloning rpclib because already exists a folder called
 "rpclib-src".
    -[install_rpclib]: Generating build...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "E:/carla/Build/rpclib-src/build/CMakeFiles/CMakeOutput.log".
See also "E:/carla/Build/rpclib-src/build/CMakeFiles/CMakeError.log".

    -[install_rpclib]: [CMAKE ERROR] An error ocurred while executing the cmake.

    -[install_rpclib]: [CMAKE ERROR] Possible causes:
    -[install_rpclib]:                - Make sure "CMake" is installed.
    -[install_rpclib]:                - Make sure it is available on your Window
s "path".
    -[install_rpclib]:                - CMake 3.9.0 or higher is required.
    -[install_rpclib]: Exiting with error...

So as proposed by andschwa (commented on 11 May 2017) the internal CMake version is being launched which in his case 3.72 and in my case 3.11 as found in this directory:

E:\Program Files E\Visual Studio IDE\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\doc\cmake-3.11

The workaround from andschwa  was to edit the cmake file to set the addess of the external CMake directory specifically to avoid the 3.72 in his case. I don't follow his code but using this
https://stackoverflow.com/questions/11588855/how-do-you-set-cmake-c-compiler-and-cmake-cxx-compiler-for-building-assimp-for-i
I modified the cmake list to:

cmake_minimum_required(VERSION 3.9.0)
set(CMAKE_C_COMPILER "C:/Program Files/CMake/bin/cmake.exe")
set(CMAKE_CXX_COMPILER "C:/Program Files/CMake/bin/cmake.exe")
project(CARLA)
include("Build/CMakeLists.txt.in")
add_subdirectory("LibCarla/cmake")

And this worked! Thanks for your help @iFuSiiOnzZ

medusa606, 2018-10-12:
Haha now another CMake error - please see below. Any ideas? Thanks!

   ###########
   # SUCCESS #
   ###########
   IMPORTANT!
   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in E:/carla/Build/ directory)
   You only need the ASSET PACK with all the meshes and textures.
   This script provides the assets for CARLA 0.9.0
   You can download the assets from here:
       https://drive.google.com/open?id=1FtC00CrDb7Kz5StBAwb6vqOGbzZtpROx
   Unzip it in the "E:/carla/Unreal\CarlaUE4\Content\Carla" folder.
   If you want another version, search it in E:/carla/Util\ContentVersions.txt.
-[BuildLibCarla]: [Batch params]: --server --client
CMake Error: Error: generator : Visual Studio 15 2017 Win64
Does not match the generator used previously: Visual Studio 15 2017
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a diffe
rent binary directory.
Error: could not load cache
CMake Error: Error: generator : Visual Studio 15 2017 Win64
Does not match the generator used previously: Visual Studio 15 2017
Either remove the CMakeCache.txt file and CMakeFiles directory or choose a diffe
rent binary directory.

-[BuildLibCarla]: [ERROR] An error ocurred while executing the cmake.
          [ERROR] Possible causes:
          [ERROR]  - Make sure "CMake" is installed.
          [ERROR]  - Make sure it is available on your Windows "path".
          [ERROR]  - CMake 3.9.0 or higher is required.
make: *** [LibCarla] Error 1

iFuSiiOnzZ, 2018-10-12:
Have you rename any folder? Sometimes is because of that. Delete the makefile from Build and try again, if not delete the Build folder and let's see.

medusa606, 2018-10-12:
So tried deleting the CMakeCache.txt file which proceeded the build a little further this time.

    11 Warning(s)
    0 Error(s)
Time Elapsed 00:00:43.22
-[BuildLibCarla]: LibCarla server has been successfully installed in "E:/carla/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "E:/carla/PythonAPI\dependencies"!
-[BuildCarlaUE4]: [Batch params]: --build --ue-version 4.19
ERROR: The system was unable to find the specified registry key or value.

-[BuildCarlaUE4]: [ERROR] Unreal Engine 4.19 not detected

iFuSiiOnzZ, 2018-10-12:
That is because the Unreal Engine path hasn't been found. For any reason no information has been written into the Windows register. You can reinstall it given administrator rights or launch the editor and build it from there.

medusa606, 2018-10-12:
Reinstall Unreal as Admin? Not sure what you mean here. There is no context option to run the .msi file as admin.

iFuSiiOnzZ, 2018-10-13:
Run a `cmd` as admin and install it from there, or just try to reinstall normal, at some point it should ask administrator rights as it needs to write the Windows register.

closed 2018-12-19 by stale[bot]