carla-simulator/carla #865: Carla WeatherId

dhernandez0, 2018-10-11:
I'm using carla 0.8.2 (stable version), I modify manual_control.py (line 81=5, MidRainyNoon), and it is not raining in the simulator, see attached image. 
[image]

marcgpuig, 2018-10-15:
Hi @dhernandez0! how are you? :)
So you are doing:
```py
WeatherId=5,
```
and is not raining, right?
Did you changed anythig from the client? Make sure you are calling `scene = self.client.load_settings(self._carla_settings)` inside `_on_new_episode()`.
Also, did you deleted `self._carla_settings.randomize_weather()`?

Cheers!

dhernandez0, 2018-10-15:
according to https://carla.readthedocs.io/en/latest/carla_settings/#weather-presets

5 is MidRainyNoon, then rainy is not rain? but likely to rain?

Oh! I didn't remove "self._carla_settings.randomize_weather()" that could be the problem, I'll check that

marcgpuig, 2018-10-15:
@dhernandez0 Acording to @TheNihilisticRobot, "Rainy" is a bit of rain.

dhernandez0, 2018-10-15:
It works now.was because of the "self._carla_settings.randomize_weather()" line. Thanks 👍

closed 2018-10-15 by dhernandez0