carla-simulator/carla #867: Build Python API

Sakrust, 2018-10-12:
I am kind of using a virtual environment for my python packages (python 2.7). But if I just run the make PythonAPI command, the .egg file is for python 3.6 (global python version). How can i specify a different python version?

iFuSiiOnzZ, 2018-10-12:
What operating system do you have? When PythonAPI is build is taking the default version of python that finds on the environment path.

For Linux the most easier way of doing is to to make an alias into the `.bashrc`, `.profile`, etc file (depends on en environment) (`echo 'alias python="/usr/bin/pythonX.X"' >> ~/.bashrc`). Another way of doing it is use the `update-alternatives`, but you will have to set it if is not already set.

On Windows make sure that path to python2.7 is set on the environment path.

Sakrust, 2018-10-12:
Win10. Thanks for your quick answer.

iFuSiiOnzZ, 2018-10-12:
Unfortunately on Windows it doesn't compile for 2.X, It seems that the build tools for python is using VS2010 build tools and it doesn't support C++11.

Sakrust, 2018-10-12:
Okay, then i'll try to update my python env.