carla-simulator/carla #872: a simple way to generate 2d bounding box?

1453042287, 2018-10-13:
there is a method to generate an image containing CARLA semantic segmentation labels, why can't we utilize the similar way to generate the 2d bounding box?
        any suggestion? thanks a lot!

analog-cbarber, 2018-10-13:
Doesn't seem that would work very well if there are overlapping objects with same segmentation. It would work better if we had an image with per-pixel agent-id labels.

It really is better to just project the 3D OBB onto the camera perspective and cull boxes that are outside the camera frustrum or behind the reported depth (requires depth images of course).

1453042287, 2018-10-15:
Thank you for the suggestions @analog-cbarber 
but actually, what i mean is not to draw the bounding box from the segmentation image, i wonder why can't we just utilize the method that generate the segmentation camera image to directly generate the 2d box?
BTW, can you briefly describe the process about how the segmentation labels is generated or some hints to the code?
thanks a lot!

closed 2018-12-21 by stale[bot]

giladdiv, 2019-01-06:
@1453042287 
Have you managed to draw 2D boxesw?

giladdiv, 2019-01-07:
@1453042287 
Thanks, but I did not understand when def _on_render(self) is being called
and how can I change the camera.listen function to draw the bbox on the images?

1453042287, 2019-01-07:
the _on_render(self) function is defined in the manual_control.py, but you can ignore that and just utilize the variable and function mentioned in the #772 , it's same to use this method in other .py file, like client_example.py and so on

giladdiv, 2019-01-10:
@1453042287 
Which version are you running? I tried 0.92 and 0.84 and it seems that the notations from the code do not exists like 
xxxx.transform
xxxx.matrix

1453042287, 2019-01-13:
@giladdiv i use the 0.82(stable) and the 0.84(have bike), xxx.transform is the measurements which mentioned in the carla's instructions, and U can find the xxx.matrix in the carla_0.8.4/PythonClient/carla/transform.py

marcgpuig, 2019-01-14:
Hi @1453042287 glad to see you are making it work!
Maybe the issue #76 can give you some clue about the instance semantic segmentation detector.
Good luck!

1453042287, 2019-01-15:
@marcgpuig thanks

geekhut, 2019-04-15:
@1453042287 Hi could you please share the code for 2D bounding box? Thanks.

1453042287, 2019-04-16:
i'm now working on the point cloud and the code for 2d bbox is not used anymore for a while and the version changes a lot , just look at the #772 and you will find a simple way to draw the 2d bbox :)

stevelaclasse, 2020-02-11:
check this issue

https://github.com/carla-simulator/carla/issues/2325