carla-simulator/carla #873: Controlling player vehicle and getting its feedback

praveenputhettu, 2018-10-15:
How can i control the player vehicle using our-own AI?
Also how i get the feed-back i.e, current steering-angle/wheel-angle, current speed/acceleration,  from player vehicle?

CARLA version number : 0.8.4
Platform/OS : Ubuntu 16.04

SpakMiguel, 2018-10-15:
In the client_examply.py file there is a print_measurements functions that prints some information of the simulation. You can modify it as you want in order to print or get the data you are interested in.
In the same file mentioned above, there is a section that deals with autopilot. It is easy to find thanks to the comments.

praveenputhettu, 2018-10-16:
Thanks for the reply SpakMiguel. 

I have gone through the control signals and player measurements signal.
Carla provides information on current pose, acceleration, forward speed, collision, lane intersections etc. 
I would like to know whether CARLA provides information on current steering angle or wheel angle of the player vehicle ?

Romanenko-Serhii, 2018-10-16:
@praveenputhettu 
current steering angle:
measurements.player_measurements.transform.rotation.roll
measurements.player_measurements.transform.rotation.pitch
measurements.player_measurements.transform.rotation.yaw

Romanenko-Serhii, 2018-10-18:
@praveenputhettu Did you see this page https://carla.readthedocs.io/en/0.8.4/measurements/ ?

praveenputhettu, 2018-11-23:
Thanks Romanenko-Serhii

closed 2018-11-23 by praveenputhettu

HubFire, 2019-01-22:
> Thanks Romanenko-Serhii

How do you get the  feed_back steer  angle ,i think  the yaw and autopilot_control is  not I  need.