carla-simulator/carla #874: Is there any way to control the speed of a pedestrian?

li27, 2018-10-15:
I want to let pedestrians walk faster, but I can't find a way to achieve it.  Is it possible?  Thank a lot.  The version I'm using is CARLA 0.8.4.  Only need to control the speed.

TheNihilisticRobot, 2018-10-19:
Hi @li27 
Sorry for answering so late.

You can't change the pedestrian speed yet on this version but we are working on absolute control over any actor from the client for the future.

For now, what you can do if you have the full project is change the Maximum and minimum speed our pedestrians spawn with. To do so you must go to our pedestrian class: (Content\Carla\Blueprints\Walkers\BP_Walker) And change the following:

- In the "CharacterMovement" component under "Character movement: Walking" Set the max walk speed as your desired number, this defines the speed when the pedestrian is carrying no prop at all.

- In the "WalkerProps" variable you'll have a series of entries that defines many things about any of the props our pedestrians carry, between them is the maximum and minimum speed they spawn with. 

With those things changed any pedestrian spawned should be walking at the speed you've set.

TheNihilisticRobot, 2018-10-26:
@bjane123 
I'm sorry but that's not the file you need to change. Those are just the default parameters of the WalkerProrps Variable. the file you need to change is Content\Carla\Blueprints\Walkers\BP_Walker.

closed 2019-01-01 by stale[bot]

YDSh510, 2019-05-24:
Hi @TheNihilisticRobot - How to add hand movements in walker . ( Moving hands up and down)
As I don't see any option via CharacterMovement component. How will I change BP_Walker to add the required logic.

YDSh510, 2019-05-27:
@TheNihilisticRobot  - Need some urgent pointers to proceed further. Please support.