carla-simulator/carla #876: Turn traffic signals into green

SpakMiguel, 2018-10-16:
I want to record RGB 10-30 seconds videos. The thing is all traffic lights are almost always red and the car spends half of the video waiting in front of one of them. I have been exploring the python client a bit and have not found a way to change traffic signal state. Only some references at carla_server_pb2.
I would appreciate if someone could help me with that.

Thanks!

TheNihilisticRobot, 2018-10-18:
Hi @SpakMiguel 

Unfortunatelly, though we do have plans to implement control over traffic lights from the client in the future, we don't yet have a clear date for when that will be an option. If you have the full Unreal project you might want to consider setting up your scenario Into the editor; Issues #648 and #515 might help you with that.

closed 2018-12-24 by stale[bot]

shh1v, 2022-07-15:
Is there a way to implement this now?

paulerikf, 2022-11-15:
@shh1v https://carla.readthedocs.io/en/0.9.13/core_actors/#traffic-signs-and-traffic-lights

shh1v, 2022-11-15:
Thanks @paulerikf ! I figured it out earlier.

soumya997, 2022-11-28:
This is a solution. I was stuck in that previously.

```python3
list_actor = world.get_actors()
    for actor_ in list_actor:
        if isinstance(actor_, carla.TrafficLight):
             # for any light, first set the light state, then set time. for yellow it is 
             # carla.TrafficLightState.Yellow and Red it is carla.TrafficLightState.Red
            actor_.set_state(carla.TrafficLightState.Green) 
            actor_.set_green_time(1000.0)
            # actor_.set_green_time(5000.0)
            # actor_.set_yellow_time(1000.0)
```

For more reference see this [carla_dataset_tools](https://github.com/KevinLADLee/carla_dataset_tools/blob/afc43efc9f18472b726261e6a5e9a2aadd3b2c4e/data_recorder.py#L48).

Nitro60zeus, 2023-08-29:
> This is a solution. I was stuck in that previously.
> 
> ```python
> list_actor = world.get_actors()
>     for actor_ in list_actor:
>         if isinstance(actor_, carla.TrafficLight):
>              # for any light, first set the light state, then set time. for yellow it is 
>              # carla.TrafficLightState.Yellow and Red it is carla.TrafficLightState.Red
>             actor_.set_state(carla.TrafficLightState.Green) 
>             actor_.set_green_time(1000.0)
>             # actor_.set_green_time(5000.0)
>             # actor_.set_yellow_time(1000.0)
> ```
> 
> For more reference see this [carla_dataset_tools](https://github.com/KevinLADLee/carla_dataset_tools/blob/afc43efc9f18472b726261e6a5e9a2aadd3b2c4e/data_recorder.py#L48).

Where do i apply this code snippet? scenario_runner.py?

Nitro60zeus, 2023-08-29:
Hi! Thanks a lot for your response. My objective is to reduce the wait time at the red lights (currently its 1min and I want to reduce it to 20 seconds).
I did this in `route_scenario.py`. I added the func block in the script, but it didn't work. Pretty sure I am doing something wrong:
[image]

shh1v, 2023-08-29:
Oh, okay. I apologize for misunderstanding your requirements. If all you want is to reduce the wait time of the traffic light, and there's no specific point in the execution where it needs to be done, you can simply define the function and call it in the end of `_load_and_wait_for_world()` of `scenario_runner.py`.

Nitro60zeus, 2023-08-29:
Ah. 
So I defined the function and call it at the end of `_load_and_wait_for_world()` . I have a small question. 
When calling, what argument will we pass for `traffic_light_setting` ?

Like: 
```
def _load_and_wait_for_world(self, town, ego_vehicles=None):
        "
        "
        //initial code
        

        set_traffic_light_time(self, ??) // <- I am trying to call here. But I am confused with the 2nd argument
         
        # Wait for the world to be ready
        
        if CarlaDataProvider.is_sync_mode():
            self.world.tick()
        else:
            self.world.wait_for_tick()

        map_name = CarlaDataProvider.get_map().name.split('/')[-1]
        if map_name not in (town, "OpenDriveMap"):
            print("The CARLA server uses the wrong map: {}".format(map_name))
            print("This scenario requires to use map: {}".format(town))
            return False

        return True
```

shh1v, 2023-08-29:
Define your function as below:
```
def set_traffic_light_time(duration=20):
    actor_list = self.world.get_actors()
    for actor_ in actor_list:
        if isinstance(actor_, carla.TrafficLight):
            actor_.set_red_time(duration)
```

Nitro60zeus, 2023-08-30:
YES! It works! Thanks a lot brother!! Really grateful for your help <3