carla-simulator/carla #885: Can Laser radar provides laser intensity information?

chengjinjack, 2018-10-18:
Hello,everyone!
I wander that if Laser radar can provides laser intensity information?  If there is a way to get  laser intensity information,how can I do it?

analog-cbarber, 2018-10-18:
The Lidar model simply reports ground-truth measurements of pixel ranges without the kind of errors you would get with a real Lidar. There is no real concept of signal intensity. You would have to provide that yourself. The easiest thing to do would be to scale the intensity by the distance squared but ideally the material properties should also be considered.

s894330, 2018-12-19:
hi
according issue 327's reply: "the easy way to simulate lidar intensity is simulated in Python in the client-side, just adding an extra layer on top of API"

May I ask how exactly should I do to let the lidar data can has intensity information? 
thank you

closed 2019-02-24 by stale[bot]