carla-simulator/carla #888: Where is the Autopliot file(s) ?

JbAmade, 2018-10-18:
Hello,
I'd like to know which file is coding the autopilot. I've read that the python client controls the simulator by driving the car, but I can't find any script or piece of code which codes the Autopilot's behavior. 
I suppose there is a script in the client which contains all the Autopilot functionalities.
Could anyone tell me where is this file(s) ?
Thanks

marcgpuig, 2018-10-19:
Hi @JbAmade.
The autopilot is done in the server side :)

JbAmade, 2018-10-19:
Really ?!
I wanted to extract the autopilot to make it drive other vehicles: other simulators, an RC car... But if the autopilot requires other inputs than just the main camera (lidars, laser...), then it may not work...

marcgpuig, 2018-10-20:
We are currently working in sending all the road information to the client, so you will be able to make your own autopilot there!
This should be ready in 0.9.1.

JbAmade, 2018-11-06:
Hello @marcgpuig !
I’ve just seen that Carla 0.9.0 has been released ! The functionalities you told me about, are they included in this version ?

My idea is that I’d like to use the autopilot to make it drive other vehicles, not only the simulator : other softwares, games, or even an RC car. Of course, as you told me, the autopilot has to be in the client. 
Let me know please
Thank you

marcgpuig, 2018-11-26:
Hi @JbAmade

You now have the OpenDrive information of the road, alongside with an API that can generate waypoints given this OpenDrive. You can make your own autopilot based on this information, for example.

Cheers

JbAmade, 2019-02-15:
Ok. Thank you for your reply and your help

closed 2019-02-15 by JbAmade