carla-simulator/carla #889: Import Google map as Carla map

hsienyan, 2018-10-19:
Hi 
I would like ask if it is possible to import 3D google map into Carla as map?

If it is possible,please explain.

Thank you.

marcgpuig, 2018-10-19:
Hey @hsienyan. 
Sorry but this is not currently possible. Some people is working with OpenStreetMaps and we are currently working on generating the traffic from OpenDRIVE. These are tools that will able you to do so, but using OpenStreetMaps.

hdy117, 2018-10-24:
@marcgpuig when will OpenDrive be supported? do you guys have a road map?

marcgpuig, 2018-10-24:
Hi @hdy117 
This will be featured in 0.9.1 that should come out in a week hopefully.
We are generating the map alongside with OpenDRIVE using [RoadRunner](https://www.vectorzero.io/).
The idea is that, in your client, you will be able to access to OpenDRIVE data itself or use a basic API that we are currently designing, to get waypoints, nearest points, and so on.

MohamedAbdelrehem, 2023-10-08:
Is there any news on uploading a map from Google maps or any other Map to CARLA for simulation