carla-simulator/carla #891: Add an int data to measurements and send to client

FarzanehJafari, 2018-10-20:
Hi,
I wanted to add an int data to measurements in carla_player_measurements structure in carla_server.h and set it in CarlaEncoder.cpp. I added bellow line in client side, 
Test = measurements.player_measurements.Test,
but I had an error that AttributeError: 'PlayerMeasurements' object has no attribute 'Test'.

Could you give me more information about client server structure?
Thanks a lot,

nsubiron, 2018-10-20:
You need to replicate the changes in all the intermediate structures too, from _"carla_server.h"_ to _"carla_server.proto"_, including the encoders in between. Here you can find a bit more info about the [design of CarlaServer](https://carla.readthedocs.io/en/0.8.4/carla_server/).

FarzanehJafari, 2018-10-20:
l studied this page about carla server structure before sent email for you, I checked intermedidate structure. but I didnt understand which lines need to change.
I check that again. Thanks

FarzanehJafari, 2018-10-21:
Hi,
Which classes/functions need to change exactly in Util->CarlaServer folder? I'm really confused. I checked all of the classes in this folder, but I didn't find any functions and I can't add my parameter. what's your recommend?

nsubiron, 2018-10-21:
Hi @FarzanehJafari,

Try searching for `forward_speed` for instance, and replicate what it's done with it. I think it's encoded twice, once to be passed to the struct at carla_server.h, and another to be copied to protobuf format. Define it too in the protobuf file carla_server.proto and run Protoc.sh, the protobuf compiler will generate code both for C++ and Python.

FarzanehJafari, 2018-10-21:
1. I added a float parameter 'test' to  'carla_player_measurements' in carla-server.h
[image]
2. Then, I added it in the CarlaEncoder.cpp
[image]
3. I changed carla-server.proto, but I didn't understand these equals (Control autopilot_control **= 10**)
[image]
4. Downloaded win-bash and installed it and ran bellow line in the command prompt
[image]
5. I ran manual_control.py in command prompt, but appeared this error
[image] 
Are you have other recommend?
Thanks,

nsubiron, 2018-10-22:
Hi @FarzanehJafari,

There is a Protocol.bat script for Windows, maybe that solves the issue. The error in Python you show is because the protoc script didn't generate the variable you added. Check out the output of the script to see if there are any errors.

These numbers after each variable in the proto file are for serialisation, they should be unique for each variable, but otherwise it doesn't matter what you put there.

FarzanehJafari, 2018-10-22:
I download protoc.exe for windows and run it. client knows this attribute now. I set test = 800 in server but client give test = 0 always. whats this problem?
Thanks

FarzanehJafari, 2018-10-24:
Hi, 
Could you help me? My parameter in client side cant give currect value. I set this parameter test = 800 in server side, but give test = 0 always. Do you know, whats this problem?
Thanks a lot,

closed 2018-12-30 by stale[bot]