carla-simulator/carla #893: Post process material for camera

eds89, 2018-10-21:
Hello folks,

I'm new to programming on the Unreal Engine. I want to create a post process material similar to the attached picture. It consists on a black rectangle drawn on the screen in Carla. From what I researched, I have to use some sort of sub_UV function and multiply the resulting pixels by 0. However, I cannot find a function to retrieve a small portion of all UVs. Any input would be very much appreciated. Kinds regards.

[image]
.

analog-cbarber, 2018-10-22:
Wouldn't it be easier to just apply this to the generated images? Why do you need to do this in Unreal?

eds89, 2018-10-22:
I need to see how disruptive this kind of thing is to trained network
models.

analog-cbarber, 2018-10-22:
Sure, but you can still do this to the generated rgb images before passing to your network. There is no need to do it in Unreal.

eds89, 2018-10-22:
I'd like to do it in unreal, as I can train existing networks and later
compare how networks trained specifically for this perform.

analog-cbarber, 2018-10-22:
Once again, there is no reason you cannot do that MUCH more easily outside of Unreal.  What would you do if you were working with real video cameras? Would you try to build a special video camera that blocks out part of the image?

I really would not waste your time with making special post-processing materials unless it is something that relies on material properties of the scene or takes significant advantage of the GPU pipeline.

eds89, 2018-10-22:
Ok. I must do it in the complicated way, inside Unreal. If you can't help,
that's fine --- I'm hopeful somebody can.

For experienced Unreal users, this should be an easy thing to do. Fingers
crossed, someone will reply with useful information. Many thanks.

analog-cbarber, 2018-10-22:
I assume you have read the Unreal documentation on Post processing materials: https://docs.unrealengine.com/en-us/Engine/Rendering/PostProcessEffects/PostProcessMaterials

You want to create a post processing material that either passes through the SceneTexture post processing input or mask it out based on the angle between the camera direction and the direction to the pixel being rendered which you can compute from the CameraDirectionVector and the CameraVectorWS nodes.

eds89, 2018-10-22:
Hi.
I know about the need for using SceneTexture and, multiplying the range of
pixels by 0 (black). However, I cannot come to find a way to select a
smaller portion of the whole screen UVs, which would correspond to the
rectangle.

I will try the CameraDirectionVector and the CameraVectorWS nodes
alternative, though.

analog-cbarber, 2018-10-22:
I think you probably want to use ScreenPosition.

eds89, 2018-10-23:
Using ScreenPosition, to me, needs to break the UV float into its
individual components, then compare if these components are within your
boundaries. I cannot find a way to do multiple conditional branching in
Unreal. For example: If (x >= start_x and x <= end_x and y >= start_y and y
<= end_y) then 0, else 1. Zero (black) and 1 (no change) would then be
added to a multiply node.

analog-cbarber, 2018-10-23:
Use arithmetic operators in place of logical ones. If you are working with ones and zeros, multiply is equivalent to and.

marcgpuig, 2018-10-24:
Also, logical operations are really expensive in shaders, try to avoid them.

eds89, 2018-11-27:
Hi. I want to implement a post process for dirt and dust on the camera. What's the easiest way to do that? Should I add some kind of noise to the trained neural network? I can see the default industry practice is to add gaussian noise when training CNN models. It's assumed this will eventually happen. I however do not know how camera occlusion (e.g. dirt) would be dealt with. There's this paper here: https://arxiv.org/abs/1807.06172. It does not use CARLA though.

closed 2019-02-03 by stale[bot]