carla-simulator/carla #894: Error compling Carla : 'make launch' throws errors

KetkiChaudhary, 2018-10-22:
Hello,

I am trying to build Carla on Ubuntu 16.04. I have downloaded it from here: 
https://github.com/carla-simulator/carla

I followed all the steps from the documentation. However, I get stuck at the step of "make launch". I get the following error.

smartfm@smartfm-desktop:~/carla$ sudo make launch
Setup.sh: llvm-5.0 already installed.
Setup.sh: boost-1.67.0 already installed.
Setup.sh: rpclib-2.2.1 already installed.
Setup.sh: googletest-1.8.0 already installed.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server" configuration.
ninja: no work to do.
[1/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Building LibCarla "Client" configuration.
ninja: no work to do.
[1/1] Install the project...
-- Install configuration: "Client"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: ERROR: UE4_ROOT is not defined, or points to a non-existant directory, please set this environment variable.
Util/BuildTools/Linux.mk:9: recipe for target 'launch' failed
make: *** [launch] Error 2

I have Unreal Engine v 4.19 built successfully on my machine. Also UE4_ROOT is a valid environment variable.
~/carla$ printenv | grep UE4_ROOT
UE4_ROOT=/home/smartfm/UnrealEngine_4.19
The above path also exists.

Also my cmake version is 3.13.

Help on this would be appreciated.

Thanks

nsubiron, 2018-10-23:
Our script only checks if `$UE4_ROOT` is a valid directory. Make sure you spelled correctly and the environment is passed to our scripts.

AI-liu, 2018-12-17:
add 
UE4_ROOT=PATH.../UnrealEngine-4.19
in Util/BuildTools/BuildCarlaUE4.sh

nsubiron, 2019-01-17:
Make sure `UE4_ROOT` points to Unreal top directory, doing `ls $UE4_ROOT` should print this

[image]

Owen-Liuyuxuan, 2019-03-23:
This is not working also.
[image]

[image]

wootae-jeon, 2019-04-03:
Without sudo commend.

closed 2019-06-09 by stale[bot]

basavaraj1995, 2019-09-10:
`export UE4_ROOT=~/UnrealEngine_4.22`

malibay96, 2020-03-24:
I don't now why but this steps also worked for me

In carla folder I did these

`sudo su`
`export UE4_ROOT=/home/YOURUSERNAME/UnrealEngine_4.22`
`make launch`

direcf, 2020-09-14:
Change UE4_ROOT=/home/smartfm/UnrealEngine_4.19 to UE4_ROOT=/home/smartfm/UnrealEngine-4.19
I also got this problem...Fighting!

GeraldGuo7, 2021-04-29:
1.**cd** 
2.type gedit BuildCarlaUE4.sh like this
guod@guod-B460MDS3H:~$ **gedit BuildCarlaUE4.sh**
3.replace it with your specific UE path
**export UE4_ROOT=~/UnrealEngine_4.22**
4.**make launch** under your carla folder

Laudarisd, 2021-08-04:
If anyone still getting same issue:
 After editing your bash file 
```
 gedit ~/.bashrc

    # or 

    gedit ~/.profile
```
You have to apply this in your root. To do this run 

```
source ~/bashrc
```

After that run 
```
make launch
```

Hope this will help

emeve, 2021-11-05:
I have solved this problem. Changing the owner of CARLA to other users. Hopes this will help any other friends.