carla-simulator/carla #897: ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused

JoveIC, 2018-10-23:
Hi,
I'm using compiled version 0.8.2, and I'm trying to run client_example.py on my desktop,
I'm using Ubuntu 18.04.1 LTS with GeForce GT 640 graphics.

I tried to open my ports using >> sudo ufw allow - as suggested in the instruction, and I got the following status.

To                         Action      From
--                         ------      ----
2000                       ALLOW IN    Anywhere                  
2001                       ALLOW IN    Anywhere                  
2002                       ALLOW IN    Anywhere   
2000 (v6)                  ALLOW IN    Anywhere (v6)             
2001 (v6)                  ALLOW IN    Anywhere (v6)             
2002 (v6)                  ALLOW IN    Anywhere (v6)  

But then I ran client_python.py,
It kept throwing me this error : 
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused

When I checked with: >> sudo nmap -sT -O localhost
It shows that I have only one port open (and it's not any of the three I opened with ufw)

When I run >> nmap localhost -p 2000
It shows: 
PORT      STATE   SERVICE
2000/tcp  closed   cisco-sccp

I've read #378 but it didn't fix my problem, I'm a total newbie at this, Am I doing something wrong?

Thanks in advance.

nsubiron, 2018-10-24:
Just to be sure, have you tried using a different port (both simulator and client)? There might be another app using that port.

closed 2018-10-31 by JoveIC

meteorshowers, 2020-02-05:
the same error  not solved

ZhangJian322920, 2020-07-30:
I have the same error。Can anyone know why。