carla-simulator/carla #899: Post processing effects "vignette" issue

SpakMiguel, 2018-10-23:
Hi all,
I am working on a project where I need to stitch images and some post processing effects make the resulting pictures look strange. In order to solve this I have been playing with the Post Process Volume in Unreal tweaking the post processing effects. The problem is that a strange intense vignette on the upper part of the images captured by cameras appear althought effects like vignette, chromatic aberration and bloom are deactivated. See pictures below:

[image]

This does not happen on the compiled version of CARLA, so it has to be related to changing the post process effects, which is the only thing I have done to the project.

Any ideas on what is happening here?

TheNihilisticRobot, 2018-10-24:
Hi @SpakMiguel 

Does this effect continue to happen if you deactivate the post process volume?

SpakMiguel, 2018-10-24:
Yes, happens also. And that is confusing, because the only thing I have changed in the project is the Post Process Volume.

TheNihilisticRobot, 2018-10-25:
I'm sorry but I have absolutely no clue about what could be causing the issue you mention, Does it also happen in the other maps?

closed 2018-12-31 by stale[bot]

NebulousNeo, 2019-09-05:
how did you connect your pythonAPI with pygame?

NebulousNeo, 2019-09-05:
okay i found a solution, it was the fault of the .egg file in the folder there wasn't the right .egg file version (carla-0.9.5-py2.7-linux-x86_64.egg instead of should be carla-0.9.6-py2.7-linux-x86_64.egg). This problem occur because i had two carla folder the first one was the old carla version(0.9.5) and the new one(0.9.6 build from source). So i copied from folder /home/xxx/Test/carla/PythonAPI/carla/dist/ (folder of 0.9.6) the egg files to /home/xxx/yyy/zzz/Carla/CARLA_0.9.5/PythonAPI/carla/dist (this is 0.9.5 where i also execute my pythonAPI). after that it worked and i could conntect to the carla world.