carla-simulator/carla #900: weird Segmentation picture

yangzuyuanhao, 2018-10-23:
Hi,
For some research reasons, I used the Segmentation camera in this way:

imagedata = sensor_data.get('CameraRGB',None)                 
bgra_data = to_bgra_array(imagedata)
rgb_img= bgra_data[:,:,:3]      #change to rgb
cv2.imwrite("../test3.jpg",rgb_img)
                 
semsegdata = sensor_data.get('CameraSegmentation',None)
semseg_bgra = to_bgra_array(semsegdata)                 
semseg_img = semseg_bgra[:,:,:3]   #change to rgb
cv2.imwrite("../test3.jpg",semseg_img)

But when I use OpenCV to generate the image, the RGB images looks okay, but the corresponding Segmentation image becomes weird, see below:
[image]
[image]
Any one knows why? Many thanks!

yangzuyuanhao, 2018-10-23:
Figured it out, close

closed 2018-10-23 by yangzuyuanhao

Capuz93, 2018-11-14:
Hi,
I have your same problem. I'm using: 

> ./image_converter -c semseg -i ../input_folder -o ../output_folder 

to generate the correct semantic images, but it doesn't work.
How did you fix this problem?
Thanks

Capuz93, 2018-11-14:
Ok. I figured it out. It is a graphic card problem.
Close