carla-simulator/carla #901: Error with 'make PythonAPI'

nicolebd, 2018-10-23:
I have built Carla from source on Ubuntu 16.04. Whenever I try to run 'make PythonAPI', I get the following:

In file included from source/libcarla/libcarla.cpp:38:
In file included from source/libcarla/SensorData.cpp:9:
In file included from dependencies/include/carla/image/ImageIO.h:9:
In file included from dependencies/include/carla/image/ImageIOConfig.h:50:
In file included from dependencies/include/boost/gil/extension/io/png_io.hpp:35:
In file included from /usr/include/png.h:321:
/usr/include/pngconf.h:383:12: error: unknown type name '__pngconf'
           __pngconf.h__ in libpng already includes setjmp.h;
           ^
/usr/include/pngconf.h:383:21: error: cannot use dot operator on a type
           __pngconf.h__ in libpng already includes setjmp.h;
                    ^
/usr/include/pngconf.h:384:12: error: unknown type name '__dont__'
           __dont__ include it again.;
           ^
/usr/include/pngconf.h:384:28: error: expected ';' after top level declarator
           __dont__ include it again.;
                           ^
4 errors generated.
error: command 'clang-5.0' failed with exit status 1
Util/BuildTools/Linux.mk:47: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

I tried setting -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=false in the setup.py file and that just produces another error. I'm not sure how to proceed. I've cleaned and rebuilt Carla but nothing seems to work.

nsubiron, 2018-10-24:
Hi @nicolebd,

You need to install `libpng16-dev`. Last merges into master require these new dependencies

    sudo apt-get install libpng16-dev libtiff5-dev libjpeg-dev

Unlingius, 2018-12-20:
@nsubiron    I try to install libpng16-dev, it depends on libpng16-devtools - this lib conflicts with many other libs in my system, including ros. In ros I use libpng12.  Is that possible to change version in carla to 12? How difficult it will be for me?

nsubiron, 2018-12-20:
Hi @Unlingius, yes, it's a known issue, unfortunately our image library (Boost.GIL) does not support libpng12. You can apply the changes in #924 for now and save the images in TIFF.

Unlingius, 2018-12-20:
Thank you so mush! It works fine!

Unlingius, 2019-01-16:
@nsubiron  i try to compile newest release 0.9.2 and this instruction doesn't work now. Compilation crashes with that error:
/usr/include/pngconf.h:383:12: error: unknown type name '__pngconf'
           __pngconf.h__ in libpng already includes setjmp.h;
           ^
/usr/include/pngconf.h:383:21: error: cannot use dot operator on a type
           __pngconf.h__ in libpng already includes setjmp.h;
                    ^
/usr/include/pngconf.h:384:12: error: unknown type name '__dont__'
           __dont__ include it again.;
           ^
/usr/include/pngconf.h:384:28: error: expected ';' after top level declarator
           __dont__ include it again.;

How can i fix it?

Unlingius, 2019-01-16:
fixed. see extra_compile_args += ['-DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=false']  in PythonAPI/setup.py

hynkis, 2019-02-16:
Thanks, Unlingius. Changing 'true' to 'false' makes it work!

closed 2019-04-24 by stale[bot]

soulslicer, 2019-08-20:
I dont see the file seutp.py anywhere!