carla-simulator/carla #903: Any documentation on the format of 3D environment and maps? How to change it?

wmpang, 2018-10-24:
As title. If we have a 3D scene, how should I put it into CARLA?

TheNihilisticRobot, 2018-10-24:
Hi @wmpang 

If you have a full scene modelled you'll just need, for starters, to import it to unreal and add it to a clear map. [Unreal's official documentation](https://docs.unrealengine.com/en-us/GettingStarted) should help you with that.

wmpang, 2018-10-24:
As I am not familiar with CARLA, I wonder if a scene is imported to Unreal, then, what should I do to make it work with CARLA's functionalities?

TheNihilisticRobot, 2018-10-24:
So, we would love to provide the tools for adapting pre-made maps to function with carla but we are far from been able to. There are many workarounds with their pros and cons. We've already covered them in some old issues, I hope these help you: 

#603 #620 #774  #636 

Please, let me know if you further help with this.

closed 2018-12-31 by stale[bot]