carla-simulator/carla #904: Where is Color definition of semantic segmentation

materdd, 2018-10-24:
Hi!
I'm tackling for semantic segmentation. And

marcgpuig, 2018-10-24:
Hi @materdd

Do you mean the label of the ids? Please, check out the [documentation](https://carla.readthedocs.io/en/latest/cameras_and_sensors/#camera-semantic-segmentation).

materdd, 2018-10-24:
@marcgpuig 
Thank you for your reply. But i mean that it is not ids, but rgb color, like road is (128, 64,128).
Where is it defined?

marcgpuig, 2018-10-24:
Oh I see!
### 0.8.x
Check out the [image_converter.py](https://github.com/carla-simulator/carla/blob/master/PythonClient/carla/image_converter.py):
https://github.com/carla-simulator/carla/blob/46d36435d4c0b769a5de20077739e9490e50e5ad/PythonClient/carla/image_converter.py#L59-L73
### 0.9.x
Check out the [CityScapesPalette.h](https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/image/CityScapesPalette.h):
https://github.com/carla-simulator/carla/blob/46d36435d4c0b769a5de20077739e9490e50e5ad/LibCarla/source/carla/image/CityScapesPalette.h#L19-L32

materdd, 2018-10-25:
@marcgpuig

Thank you for your support. it helps me a lot.

closed 2018-10-25 by materdd