carla-simulator/carla #905: Carla ROS bridge: 'SceneDescription' object has no attribute 'map_name'

anshulpaigwar, 2018-10-24:
I am trying to use carla_ros_bridge package Carla version 0.8.4 and ros kinetic

Running the test command fails
```
rostest carla_ros_bridge ros_bridge_client.test
```

Output:
```
Traceback (most recent call last):
  File "/home/anshul/catkin_ws/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 42, in <module>
    main()
  File "/home/anshul/catkin_ws/src/carla_ros_bridge/src/carla_ros_bridge/client.py", line 38, in main
    carla_ros_bridge.run()
  File "/home/anshul/catkin_ws/src/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 143, in run
    map_handler = MapHandler(scene.map_name)
AttributeError: 'SceneDescription' object has no attribute 'map_name'
[INFO] [1540391064.654831]: Shutdown requested
[Testcase: testtestTopics] ... ERROR!
ERROR: max time [20.0s] allotted for test [testTopics] of type [rostest/publishtest]
  File "/usr/lib/python2.7/unittest/case.py", line 329, in run
    testMethod()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
    self.test_parent.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
    return self.runner.run_test(test)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/launch.py", line 684, in run_test
    (test.time_limit, test.test_name, test.package, test.type))
--------------------------------------------------------------------------------

[ROSTEST]-----------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 0
 * ERRORS: 1
 * FAILURES: 0

```

AttributeError: 'SceneDescription' object has no attribute 'map_name'

laurent-george, 2018-10-25:
Strange. Did you have a previous version of carla installed on your computer ?

MustafaIsmaill, 2018-10-25:
I also have the same exact error. I am running 0.8.2 stable version and no I had no previous versions installed on my computer.

anshulpaigwar, 2018-10-26:
I also have Carla 0.82
Here is a workaround, I hardcoded scene.map_name as string "Town01"
```
map_handler = MapHandler("Town01")
```

Then you have another error that CarlaMap object takes 4 arguments but only 2 provided.
Missing values for pixel_density, node_density
For this, you have to change the map.py file line 20:

```
self.carla_map = CarlaMap(map_name)
```
to

```
self.carla_map = CarlaMap(map_name, 0.1643, 50)
```
pixel_density =  0.1643,  node_density = 50

I found the values from previous versions of this file

MustafaIsmaill, 2018-10-26:
hmm.. I did another workaround. The following 2 lines: 
map_handler = MapHandler(scene.map_name) 
map_handler.send_map()

as mentioned in a comment they are responsible for "sending occupancy grid to rivz" and I was not planning on using rviz at the moment so I commented them and everything worked fine. Maybe I can note down your method if I plan to use rviz in the future.

danil-tolkachev, 2018-10-26:
Hi guys! Sorry, it is not documented. Working combination:
* Carla binaries form 0.8.4 release
* `PythonClient` from master branch
* `carla_ros_bridge` from master branch

MustafaIsmaill, 2018-10-26:
well thanks Danil, if I knew this info 3 days ago I would have saved a lot of effort and time :/ but thanks anyway