carla-simulator/carla #906: Python API and Sever for Carla 0.9.0

aroongta, 2018-10-25:


nsubiron, 2018-10-25:
Hi @aroongta,

Yes, sorry about the lack of documentation, we'll work on that soon.

I'd recommend getting the last version in `master`, as there are a few good improvements for the Python API. Do a `make clean` after `git pull` btw.

To run the Python examples inside `PythonAPI` you need to compile the Python package.

    make PythonAPI

after that you should be able to execute the Python scripts with Python 2.7.

aroongta, 2018-10-25:
Hi @nsubiron 
Thank you so much for the help. I will switch to the latest version in master and try to figure it out. If possible, please share a mail id/ contact where I can contact you for further advice.
Thank you

harlowja, 2018-10-25:
I would jump on discord and try the #development channel. I have gotten it to work (and others as well) so can likely help out.

aroongta, 2018-10-26:
Thank you so much @harlowja 
I would definitely do that.

genipap, 2018-10-31:
make PythonAPI which works but make check.PythonAPI didn't work. Error is 

ERROR: test_client_version (test.test_client.testClient)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_client.py", line 17, in test_client_version
    c = carla.Client('localhost', 8080)
AttributeError: 'module' object has no attribute 'Client'

======================================================================
ERROR: test_default_values (test.test_transform.testLocation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_transform.py", line 14, in test_default_values
    location = carla.Location()
AttributeError: 'module' object has no attribute 'Location'

======================================================================
ERROR: test_named_args (test.test_transform.testLocation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_transform.py", line 32, in test_named_args
    location = carla.Location(x=42.0)
AttributeError: 'module' object has no attribute 'Location'

======================================================================
ERROR: test_default_values (test.test_transform.testRotation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_transform.py", line 52, in test_default_values
    rotation = carla.Rotation()
AttributeError: 'module' object has no attribute 'Rotation'

======================================================================
ERROR: test_named_args (test.test_transform.testRotation)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_transform.py", line 70, in test_named_args
    rotation = carla.Rotation(pitch=42.0)
AttributeError: 'module' object has no attribute 'Rotation'

======================================================================
ERROR: test_print (test.test_transform.testTransform)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_transform.py", line 113, in test_print
    t = carla.Transform(
AttributeError: 'module' object has no attribute 'Transform'

======================================================================
ERROR: test_values (test.test_transform.testTransform)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_transform.py", line 90, in test_values
    t = carla.Transform()
AttributeError: 'module' object has no attribute 'Transform'

======================================================================
ERROR: test_default_values (test.test_vehicle.testVehicleControl)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_vehicle.py", line 14, in test_default_values
    c = carla.VehicleControl()
AttributeError: 'module' object has no attribute 'VehicleControl'

======================================================================
ERROR: test_named_args (test.test_vehicle.testVehicleControl)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/nsh1609/Desktop/carla/PythonAPI/test/test_vehicle.py", line 28, in test_named_args
    c = carla.VehicleControl(
AttributeError: 'module' object has no attribute 'VehicleControl'

closed 2019-01-06 by stale[bot]