carla-simulator/carla #911: problem to implement a new feature to get actor by its id for version 0.9

matthew-rococo, 2018-10-26:
First to forgive my poor English, I will try my best to make it clear. 
Version 0.9 is a very big step for Carla system. I tested it for many times and found out maybe a new feature is needed:  one client connect to the server, doing some test and finished, but don't destroy the vehicle it spawned. a new client connected to the server ,it can control the vehicle of the first client.
I carefully studied the server code and didn't find this function. so I write it by myself . I am not good at c++ and python, but  after some hard works, I made it except with some strange things.  
I added the following code to the file \CARLA\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Server\TheNewCarlaServer.cpp in the BindActions() method:
`  Server.BindSync("get_actor_by_id", [this](int id) -> cr::Actor {
    RequireEpisode();
    auto ActorView = Episode->GetActorRegistry().Find(id);
    if (!ActorView.IsValid() || ActorView.GetActor()->IsPendingKill()) {
      RespondErrorStr("unable to get actor: actor not found");
    }
    return SerializeActor(ActorView);
  });`

 sure, I also add some other code at the other related files , but I don't think it is important to this problem because this function is almost worked,  so I won't paste the other codes here.

  This function work!  In python command line, i  make an connect to the server ,get the world, spawn a new car, do some operation to the car, set_location,  set_autopilot, etc, it works fine . then i write down the car's id(for example, its id is 3) and disconnect from the server by close my python procedure. 

again, I connect to the server and get the world, this time I doesn't spawn a new vehicle, instead, I use my new function like this : my_car = world.get_actor_by_id(3),   when I do my_car.set_location(somelocation) , my_car.get_location(), it is OK and the car actually moved. but when I do like this : my_car.destory()  my_car.set_autopilot(True) . it returned error said "AttributeError: 'Actor' object has no attribute 'destory' " or  "AttributeError: 'Actor' object has no attribute 'set_auto_pilot' ".   

The command I first issued at the python command:

[image]

The command I second issued at the python command:
[image]

matthew-rococo, 2018-10-28:
I pulled the new version and can't build it on my windows system( the old 0.9.0 version can build without problem). i try to change some codes as the following:
1) LibCarla\source\carla\client\detail\ActorState.h

@@ -44,7 +44,7 @@ namespace detail {
    World GetWorld() const {
      return World{_episode};
    }

    
  protected:

    const rpc::Actor &GetActorDescription() const {
@ -59,7 +59,7 @@ namespace detail {
      return _episode;
    }

  private:
  public:

    friend class Simulator;

2)LibCarla\source\carla\geom\Location.h

@@ -28,6 +28,8 @@ namespace geom {
    using Vector3D::msgpack_unpack;
    using Vector3D::msgpack_object;

	Location() = default();

    Location &operator+=(const Location &rhs) {
      static_cast<Vector3D &>(*this) += rhs;
      return *this;

but still there are a lot of problems and seems they are related to UE4 Engine code,  i am completly confused, the following is a picture of the compiling problem log:
[image]
Help me, please!!!

nsubiron, 2018-10-29:
Hi @matthew-rococo, yes, I can reproduce. Unfortunately we are super busy now with the 0.9.1so I'm afraid support for Windows will have to wait :( sorry about that

matthew-rococo, 2018-10-30:
> Thanks for your patient. I have found a temporary solution for this problem in another issue and now I can run Carla in the Editor mode, but when I compile the PythonAPI, there are new errors occurred, I correct some of it and still there are two, it seems that you have forgot to upload two include files: Memory.h and PythonUtil.h :
>
> building 'carla.libcarla' extension
> C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Users\WangLi\AppData\Local\Programs\Python\Python37\include -IC:\Users\WangLi\AppData\Local\Programs\Python\Python37\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17134.0\cppwin [... cut ...]
> source/libcarla/libcarla.cpp(7): fatal error C1083: Cannot open include file: 'carla/Memory.h': No such file or directory
> error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
>
> -[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:/CARLA/PythonAPI\dist"!

nsubiron, 2018-10-30:
That's probably CMake and its cache...   try doing a clean build

    make clean PythonAPI

but anyway it's going to fail due to the dependencies on image formats. For now, maybe you can build the client-side on a Linux environment and use it from there.

closed 2019-01-05 by stale[bot]