carla-simulator/carla #917: Runnig CARLA on cloud (AWS EC2)

AliBaheri, 2018-10-26:
Hi,

Dose anyone run CARLA on AWS EC2. How can I get CARLA run on AWS? Is there any toturial or documentation in this regard. The old issues did not help a lot.

Best,

analog-cbarber, 2018-10-26:
Yes, you can run it on EC2. I don't think there are explicit instructions for that but there is this:
https://carla.readthedocs.io/en/stable/carla_headless/

closed 2019-01-01 by stale[bot]

wendersonj, 2019-02-13:
Hi ! I want to reopen this issue. I'm trying to use the tutorial to carla_headless in the latest (0.9.3), but everytime I get stucked in `sudo nohup Xorg :7 & ` it says that the command Xorg doens't exist ! Could anyone help me ?

Last lines of log:
[image]

[Full Log](https://pastebin.com/7XYt7n2X)

thefatbandit, 2020-06-23:
Thank Jbnunn https://github.com/jbnunn/CARLADesktop.
Succesfully run carla on AWS EC2.

jbnunn, 2022-09-15:
I'll be updating this soon to run with v 0.9.13