carla-simulator/carla #920: Split off ros-bridge into carla-ros repo

harlowja, 2018-10-27:
IMHO it should be its own repo with its own requirements and its own tests and...

I can even (maybe) contribute our ros usage as a starting point (and work to integrate the other ros functionality that exists into that).

laurent-george, 2018-10-30:
Could be a good idea (it will allow to track issues related to ros bridge more easily in my opinion).

nsubiron, 2018-10-30:
We will discuss it here with the team to see all the pros and cons, but yeah IMO feels a bit disconnected from the rest of the repo, seems like it deserves its own.

harlowja, 2018-10-30:
I can also figure out how to contribute a dedicated ros<->manual_control(like) program that I am building/working on that should I think act as a good collaboration area for all ros folks (right now most of the ros folks I think are POCing with 0.8, which I am not doing, I am using 0.9 even though that moves more). When that happens it starts to feel like it deserves its own repo and such (though that may be more of a PITA than its worth, idk).

danil-tolkachev, 2018-10-31:
It sounds reasonable. It would also be a more convenient way to install ros bridge without symlinks.

PR #1007 (2018-11-30): Move ROS bridge to its own repo
As for #920, it seems better to have a dedicated repository for the Carla ROS bridge.

Closes #920.

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/1007)
closed 2018-11-30 by nsubiron