carla-simulator/carla #932: Running client_example.py cause Unreal crashed

Johnnyswong, 2018-10-31:
Hi, I can run Carla on my server and successfully run a carla server, but when I click play and running client_example.py, Unreal is always crashed... 
I have try your driving_benchmark.py, the problem is that the code can't connect to the carla server with default port 2000. Is it the port 2000 on my server don't work? How to setup world port in Carla?

nsubiron, 2018-10-31:
It may be related to #928.

Johnnyswong, 2018-11-01:
Thx, @nsubiron  but can you tell me where is the rpclib directory? I can't locate in my server.

Johnnyswong, 2018-11-01:
I have read your script in Util/BuildTools/Setup.sh, does it means that I have to pull your new 0.8.1 version and reinstall Carla?

Johnnyswong, 2018-11-01:
After I reinstall rpclib in Carla/Build and delete the previous version, the carla server still crashing when the carla client connecting to it. My server is Ubuntu 16.04 and the script I run is client_example.py.

Johnnyswong, 2018-11-01:
In addition, I find a new bug when I am running the script PythonClient/manual_control2.py. The error is "ALSA lib confmisc.c:768:(parse_card) cannot find card '0'", is that means I should install a dummy sound card?

nsubiron, 2018-11-02:
Hi @Johnnyswong,

Sorry, I misread your description, it seems you're using a version prior to 0.9.0, it that right? then it's not related to #928.

Which version exactly are you using? Are you using the binaries we released or you built from source?

Johnnyswong, 2018-11-06:
Hi @nsubiron 
I am using the 0.9.0 version. I have try binaries and source code version, but both of them didn't work.

nsubiron, 2018-11-06:
Hi @Johnnyswong,

Please pull the last version from our master branch, and recompile
```sh
git pull
./Update.sh
make package
```

That generates a packaged version with all you need.

Johnnyswong, 2018-11-13:
Thx. I have pull the last version. But it has conflict in carla/client/Client.cpp, conflict information is: 

"CONFLICT (modify/delete): LibCarla/source/carla/client/Client.cpp deleted in 4ca95b7e88a0b94591d0874c02a30c50e6a86715 and modified in HEAD. Version HEAD of LibCarla/source/carla/client/Client.cpp left in tree."

How to fix this issue?

Johnnyswong, 2018-11-13:
By the way, I have installed carla on my server which it can't update the content directory with Update.sh. Can you provide another way to update the content directory?

Johnnyswong, 2018-11-13:
@nsubiron  I downloaded content.zip from google drive in my previous installation, and I think the error of "Error: CDO Constructor (CarlaMapGenerator): Failed to find StaticMesh'/Game/Carla/Static/RoadLines/St_xxxx" is caused by non-full update of the content files.

Johnnyswong, 2018-11-13:
Hi @nsubiron 
when I run the above commands, I meet the following error:
"x86_64-linux-gnu-gcc: error: /data/huangjunning/carla/PythonAPI/dependencies/lib/libboost_filesystem.a: No such file or directory
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
"

nsubiron, 2018-11-13:
Hi @Johnnyswong,

I see what's the issue, we require now boost_filesystem but our script does not recompile boost if the folder is already there. Ok, let me update the steps for upgrading Carla
```
git pull
./Update.sh
rm -Rf ./Build/boost-1.67.0-install/
make clean package
```
That will take a while but should work!

Johnnyswong, 2018-11-14:
Hi @nsubiron 

I follow your instructions above, but I got:

gcc: error: /data/huangjunning/carla/PythonAPI/dependencies/lib/libboost_python36.a: No such file or directory
error: command 'gcc' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

nsubiron, 2018-11-14:
Hi @Johnnyswong,

I think you are still missing dependencies, and I also realised that you're using gcc instead of clang.

Just do a clean install of Carla following all the steps in the documentation, https://carla.readthedocs.io/en/latest/how_to_build_on_linux/, that's probably the easiest.

Johnnyswong, 2018-11-15:
I follow the steps in the documentation except I run the Update.sh with flag -s, but I download the content and put it into ./Unreal/CarlaUE4

But I got the following bug:

ERROR: UBT ERROR: Failed to produce item: /data/huangjunning/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
Total build time: 98.72 seconds (Local executor: 0.00 seconds)
Makefile:620: recipe for target 'CarlaUE4Editor' failed
make[1]: *** [CarlaUE4Editor] Error 5
make[1]: Leaving directory '/data/huangjunning/CARLA/Unreal/CarlaUE4'
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

nsubiron, 2018-11-15:
Hi, the Content should be downloaded into ` ./Unreal/CarlaUE4/Content/Carla`.

Johnnyswong, 2018-11-15:
Hi, I have place the Content into ./Unreal/CarlaUE4/Content/Carla. But still not working. 
All the dependencies are the same as the tutorial recommends. 
I will try another installation from scratch. 
Has the compiled version code fixed the rpc bug?

nsubiron, 2018-11-15:
Not yet, we'll have a new binary today or tomorrow. You can also download nightly builds from our Jenkins, e.g. http://158.109.9.218:8080/blue/organizations/jenkins/carla/detail/lane_detector/16/artifacts/

Johnnyswong, 2018-11-19:
I try to install carla within the source code version and meets the error 
"make[1]: *** No rule to make target 'CarlaUE4Editor'.  Stop."

And the compiled version Carla  always crashed when the client try to connect to the server...

Johnnyswong, 2018-11-19:
BTW, can you provide the ranges of  the sensors and controls? 
I can locate these information from your documentation

Johnnyswong, 2018-11-20:
@nsubiron  And how to make sure "make launch" is success

closed 2018-11-20 by Johnnyswong