carla-simulator/carla #935: how to customize map with different curvature?

18605973470, 2018-10-31:
Curves in the original map are near 90 degrees, which are far from reality. So I want to customize it.

nsubiron, 2018-11-02:
Hi @18605973470,

I'm sorry, I'm afraid this map requires the changes introduced in later versions, I'm not sure it would work with 0.8.2.

Maybe you can try opening the UE4 editor on our last version, exporting the map, and re-importing it in 0.8.2. But there's going to be a bunch of issues, some missing C++ classes and problems with downgrading UE4.19 map to UE4.18.

quanheng123, 2018-11-02:
Hey, Generating customized map in UE4 in CARLA 0.8.2 with lane informations are the tasks I recently working on , I think you should check the picture in my issue #825 ,if that is what you want,I can give you a little advise:
First you have to create static meshes that you want to , these static meshes are the parameters that you can combine to become a road with curvatures. Second add these static mesh in Carla  source files (carlamapgenerator and so on)and also add the lane informations like the the degree of the angles of the road in roadmap file.finally rebuild Carla and you can open ue4 then drag them in editor and add them in carlamapgenerator and here you go you have a road with curvature.
Hope it helps.

closed 2019-01-08 by stale[bot]