carla-simulator/carla #937: Hallo, running client_example.py cause an exception in server.

zhangxiutao, 2018-11-01:
Hello, I am using visual Studio 2017 on windows10. Carla is the latest Version.

I can successfully run Carla Editor with the mode "Development Editor", because the mode "Development"cannot compile.

After I click run in the Editor. The Server runs without an issue.
But when I run the Client_example.py. The UE4Editor.exe throws an exception in the file 

C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\include\thr\xthread

at the following function

static unsigned int __stdcall _Call_func(void *_Data)
{	// entry point for new thread
	static_cast<_Pad *>(_Data)->_Go();
	return (0);
}

Through Debugging, I have found that, whenever the Client_example.py runs the following line:
scene = self.client.load_settings(self._carla_settings)

The Server crashes with the above mentioned exception .

What could be the Problem? Thank you very much.

zhangxiutao, 2018-11-01:
Ah in Addition the exception is

Unhandled exception at 0x00007FFA4C0DA388 in UE4Editor.exe: Microsoft C++ exception: clmdep_msgpack::v1::type_error at memory location 0x0000003D8A3BF140. occurred

nsubiron, 2018-11-05:
It could be related to #928.

yyuuliang, 2018-12-05:
I have the same trouble. The server window quit after I start example_client.py  I have already updated the new #928  code.
I am using the fix_2017_build_solution branch. Because I can't build the master branch. My pc is win10.
Any idea?

yyuuliang, 2018-12-05:
is PythonAPI for 0.9.x and PythonClient for 0.8.x ?

yyuuliang, 2018-12-05:
Ok I think I found the reason...
I am using the fix_2017_build_solution branch. I found that in the source code, the two folder "PythoAPI " and "PythonClient" both exists. I believe that PythonAPI is for 0.9.x while PythonClient is for 0.8.x. If I run script in PythonAPI, it works!

closed 2019-02-11 by stale[bot]