carla-simulator/carla #940: fatal: repository 'https://github.com/EpicGames/UnrealEngine.git/' not found

fcagroupj, 2018-11-02:
When I tried to build carla as https://carla.readthedocs.io/en/latest/how_to_build_on_linux/, I have the problems shown:
fatal: repository 'https://github.com/EpicGames/UnrealEngine.git/' not found

fcagroupj, 2018-11-02:
I found and fixed, thanks:)
https://answers.unrealengine.com/questions/422750/why-cant-i-find-repository.html

closed 2018-11-05 by nsubiron

faizan002, 2021-03-25:
I have the same issue when I go to https://github.com/EpicGames
I do not see UnrealEngine there. :(

iscanGH, 2022-01-17:
> I have the same issue when I go to https://github.com/EpicGames I do not see UnrealEngine there. :(

Did you find the solution? I have the same problem

xiaoqingnianbudadun, 2022-01-17:
I get the code here, https://github.com/CarlaUnreal/UnrealEngine
there is another repository

debipe20, 2023-05-16:
I have the same issue, could someone add me to the group?

Thanks.