carla-simulator/carla #946: Acceleration is abrupt without any control

feiyangsb, 2018-11-06:
Hello,
I am running carla 0.8.4 in Ubuntu16.04 and trying to record the acceleration in a straight lane. At first, the car accelerates to about 90km/h(with full throttle), and then close the controller(send throttle=0.0, brake=0.0, steer=0.0), recording the acceleration, [-0.4492, -0.4349, -0.4372, -0.4395, -0.4412, -0.4420, -7.5128, -7.1190, -6.6138]m/(s^2). It looks abnormal after the 6th data(it should change smoothly in my understanding). Any idea about this?
Thanks!

marcgpuig, 2018-11-26:
Hi @feiyangsb, sorry for the late response. 

Which is the interval between speeds in seconds?
When the car spawns, is from a few meters high. Maybe this _weird_ data is because the car is falling for a few simulation iterations. Also, the car's engine needs a few more simulation iterations to start.

May this be your problem?

feiyangsb, 2018-11-28:
Hi @marcgpuig . Thanks for your reply. I set the fps is 15 and what I did I accelerate the car to 90km/h(which takes some time after the car spawning) and then close the control. I think the car's engine has already started when I  measured the data.

marcgpuig, 2018-11-29:
@feiyangsb 
So, did you checked that is not an issue related to the spawn height?

feiyangsb, 2018-11-30:
@marcgpuig 
I measured the data after the car accelerate to 90km/h on the road. In my opinion, it is not related to the spawn height.

marcgpuig, 2018-12-10:
Hi @feiyangsb.
In my understanding, the acceleration should not change smoothly. 
When you accelerate full throttle, your acceleration is set to a value that will change your velocity smoothly.

feiyangsb, 2018-12-10:
Hi @marcgpuig .
Even though that, the deceleration -7.11m/s^2 is unusual when there is no brake and no throttle.

marcgpuig, 2018-12-11:
Hello @feiyangsb 
The thing is that you have a gear activated on your car's engine, this results on a deceleration.
We are working to provide gear control to change it from the Python API.
Sorry about that!

marcgpuig, 2018-12-13:
@feiyangsb 
I forgot to say that these changes are currently implemented in #1030, feel free to test that branch and see if your problem persists.

feiyangsb, 2019-01-30:
@marcgpuig 
Thanks a lot for your help. I did not try the new code but I tried to disable the gear according to #269 via adding a single very long gear. This makes it work.
Thanks again!

closed 2019-01-30 by feiyangsb