carla-simulator/carla #962: when own car model  is imported into carla,press w on the keyboard,car can't move forward and keep still,how can I  solve it?

chengjinjack, 2018-11-12:
Hello,everyone.
I have successfully implemented own solidworks 3D small car model added into the UE4,but I run into a issue,which is that when I press the play button,car can appear on the floor in the UE4,press w on the keyboard,car keep still and can't move forward .How can I solve this problem?

TheNihilisticRobot, 2018-11-12:
Hi @chengjinjack do the wheels of the vehicle reach the floor do they sink under the floor or do they stay on the air?

chengjinjack, 2018-11-14:
@TheNihilisticRobot .
HI.
Yes.The wheels of the vehicle reach the floor,but the wheels of the vehicle sink under the floor slightly,press w on the keyboard,car keep still and can't move forward.How can I solve this problem?

chengjinjack, 2018-11-14:
Hello,@TheNihilisticRobot 
Now I provide some my car's  model file,including the file(physics asset.png) in the UE4 and  the file(blender.png) in the blender.
[image]
[image]

chengjinjack, 2018-11-14:
In the simulation,car looks like as follows:
[image]

chengjinjack, 2018-11-14:
I could offer  the blender file,I hope everyone can help me see where the problem is.

[Car.blend.zip](https://github.com/carla-simulator/carla/files/2579037/Car.blend.zip)

chengjinjack, 2018-11-14:
I could offer the blender file and fbx file,I hope everyone can help me see where the problem is.
[model.zip](https://github.com/carla-simulator/carla/files/2579045/model.zip)

matthew-rococo, 2018-11-14:
You set too small shape radius in your vehiclewheel blueprint. Try to set a bigger one. Also there is another problem though it may not influence the vehicle moving: Your wheel collision is not at right position (see following pic). This may be caused by wrong scale in blender.

王立

chengjinjack, 2018-11-14:
@matthew-rococo .
Thank you!
By using the command(pxvis collision 1),I found out that own car's wheel colliders aren't on the  wheel mesh,how should I solve the issue?

chengjinjack, 2018-11-15:
Hello,everyone!
I had been solved the issue,but when stepping on the throttle, the car will lean backwards .When the throttle is released, the wheels stop moving immediately, causing the car to roll over.How should I solve it?﻿
look at the video:
https://www.youtube.com/watch?v=5Eu0gt6MP8k

TheNihilisticRobot, 2018-11-16:
Hi @chengjinjack 

Your vehicle is quite tall. The automatic physics generated by the motor might be a bit unstable. Aplying an offset to the center of mass in the Vehicle Blueprint should fix that issue. Open your vehicle blueprint clic the mesh component and look for the center of mass offset option. Tweak it until you get the desired results.

chengjinjack, 2018-11-17:
@TheNihilisticRobot 
Thank you very much!

closed 2018-11-26 by marcgpuig