carla-simulator/carla #968: Reproducing a trajectory with previously saved controls

bornabesic, 2018-11-14:
Hi,

I am doing the following:
1. Run a fixed time-step simulation (__with__ vehicles and pedestrians) using the autopilot controls. These controls are being saved for reproducing the trajectory.
2. Run the simulation again (__without__ vehicles and pedestrians), using the saved controls from the previous episode.

Basically, I am trying to have corresponding images with and without dynamic objects.
The problem I get is that the _error in the poses accumulates over time_, i.e. images that I get from the camera for the same frame are not well aligned anymore.

Example:
[image]

Any hints how to resolve this?
Thanks!

YashBansod, 2018-11-22:
@bornabesic are you running the Simulator in fixed time step mode or a variable time step mode?
I think running the Simulation in fixed time step mode may help solving this issue.

bornabesic, 2018-11-22:
@YashBansod Yes, I am running the simulator in a fixed time-step mode (`-benchmark -fps=30`).

bornabesic, 2018-11-28:
@nsubiron I am sure there is nothing that affects the car's pose in the presence of other actors (e.g. such as car crashing into another) except the autopilot controls. Wouldn't be pretty straightforward to reproduce the trajectory using the same controls (since car is not aware there are no other actors)?

nsubiron, 2018-11-28:
Using the autopilot, even if there is no collision they may reduce speed if they detect some obstacle at a distance. But if you give the exact same controls at the exact same time and use fixed time step, then yes, you should get the same poses.

bornabesic, 2018-11-28:
> But if you give the exact same controls at the exact same time and use fixed time step, then yes, you should get the same poses.

That is exactly what I am doing: exactly the same controls at the same time step; one episode with and one without other actors. But I am still having the problem I explained above and I don't know why. Poses are roughly the same, but the error between becomes too high eventually.

nsubiron, 2018-11-28:
Do you have some code that you can share to reproduce it?

bornabesic, 2018-11-29:
Sure, you can find the script I am using [here](https://gist.github.com/bornabesic/b24a149e84c4bd4223190f328645fd55). It runs two servers in parallel.

nsubiron, 2018-11-29:
I see, there are indeed small differences that accumulate, I have also tried adapting our "test_repeatability.py" script. Not sure what can be causing this.

nsubiron, 2018-11-29:
The only thing I can think of is that the physics computation is not exactly the same and floating-point inaccuracies accumulate over time.

bornabesic, 2018-12-07:
I am not excluding the physics engine error but it's weird. Sometimes I can reproduce the trajectory without a significant error whatsoever.

bornabesic, 2019-02-03:
I just got a new insight what precedes the divergence of trajectories. Every time the car stops and starts driving again, the location error just jumps all of a sudden.

Maybe this information could be helpful.

germanros1987, 2020-03-19:
This feature is now solved using the session logging and playback engine.

closed 2020-03-19 by germanros1987