carla-simulator/carla #969: Lane and poles are the same color in semantic segmentation

Capuz93, 2018-11-14:
Hi,
I obtained the images from Camera Segmentation but I discovered that "RoadLines" and "Poles" classes have the same color in the images (153, 153, 153).

How can I fix this problem?

Thanks

nsubiron, 2018-11-15:
Hi @Capuz93,

You're right! The road lines used to be (157, 234, 50). It can be modifies in [CityScapesPalette.h](https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/image/CityScapesPalette.h).

I'm not sure what's the color in CityScapes, can't seem to find road lines there. When I did that I was using the labels [here](https://github.com/mcordts/cityscapesScripts/blob/master/cityscapesscripts/helpers/labels.py).

Capuz93, 2018-11-15:
Thanks.
I changed the values in CityScapesPalette.h but the color of lanes and poles are still the same.
 What should I have to do?

Thanks

Capuz93, 2018-11-15:
Ok I fixed it. I changed label_pixel_converter.h.

Thanks

nsubiron, 2018-11-15:
Wait, don't close it until we fix it too :sweat_smile:

PR #936 (2018-11-16): Road information based on OpenDrive (WIP)
#### Description

Work-in-progress branch for the generation of waypoints based on OpenDRIVE files + a lot of tiny fixes for 0.9.1.

- Add an OpenDrive file parser.
- Add our custom map classes for easily handling OpenDrive data.
- Expose these map classes to Python API (Map and Waypoint).

###### Other fixes/changes

- Fix FileSystem to accept paths without parent folder.
- Fix several issues with the Windows compilation.
- Fix wrong map name is returned when playing in editor.
- Fix camera position in vehicle_gallery.py.
- Fix #953 UE4 circular depen [... cut]
closed 2018-11-16 by nsubiron