carla-simulator/carla #971: Vehicle spawned  in Town03 with autopilot mode keeps moves in circle

Kevin199Q, 2018-11-16:
Hello, when i spawn a vehicle in town03 using tutorial.py, the vehicle keeps moving in circle.  And it doesn't work when i spawn vehicle in other positions.

Does anyone encounter with the same situation. Thank you!

Kevin199Q, 2018-11-16:
Hi @iFuSiiOnzZ ,
Thank you for the replay. 
How could I get the certain positions in Town03 where i can spawn the autopilot vehicle with python?

Kevin199Q, 2018-11-16:
Thank you very much. I will try it.

closed 2018-11-16 by Kevin199Q

YDSh510, 2019-05-06:
@Kevin199Q and @iFuSiiOnzZ 

I tried running tutorial.py but my camera is never focus to the ego or npc vehicle and its some where else in the scene and I and not able to see any of the ego or npc vehicle spawning or moving. Do we need to do any change in script?