carla-simulator/carla #972: build on windows error  Ue4

nicksunyang, 2018-11-17:
“D:\carla\Unreal\CarlaUE4\CarlaUE4.sln”(默认目标) (1) ->
       “D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj”(默认目标) (2) ->
       (Build 目标) ->
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(59): error C2039: “_
       InterlockedCompareExchange”: 不是“FWindowsPlatformAtomics”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\
       CarlaUE4.vcxproj]
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(59): error C2665: “_
       InterlockedCompareExchange”: 4 个重载中没有一个可以转换所有参数类型 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.v
       cxproj]
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(78): error C2039: “_
       _faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxpr
       oj]
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHICommandList.h(282): error C2039:
        “_InterlockedIncrement”: 不是“FWindowsPlatformAtomics”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         d:\program files (x86)\epic games\ue_4.19\engine\source\runtime\rhi\public\RHICommandList.h(282): error C2665:
        “_InterlockedIncrement”: 4 个重载中没有一个可以转换所有参数类型 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxp
       roj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1689): error C20
       39: “GetObjectW”: 不是“TScriptInterface<IBlendableInterface>”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFil
       es\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1709): error C20
       39: “GetObjectW”: 不是“TScriptInterface<IBlendableInterface>”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFil
       es\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(3
       7): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(4
       5): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       28): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       37): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1689): error C20
       39: “GetObjectW”: 不是“TScriptInterface<IBlendableInterface>”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFil
       es\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1709): error C20
       39: “GetObjectW”: 不是“TScriptInterface<IBlendableInterface>”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFil
       es\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(3
       7): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(4
       5): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       28): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       37): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(3
       7): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(4
       5): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       28): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(1
       37): error C2660: “GetObjectW”: 函数不接受 0 个参数 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj
       ]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(103): erro
       r C2039: “__faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(154): erro
       r C2039: “__faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(159): erro
       r C2039: “__faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(165): erro
       r C2039: “__faststorefence”: 不是“FWindowsPlatformMisc”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\Car
       laUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\SlateCore\Public\Rendering\DrawElements.h(723)
       : error C2039: “_InterlockedIncrement”: 不是“FWindowsPlatformAtomics”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\SlateCore\Public\Rendering\DrawElements.h(723)
       : error C2665: “_InterlockedIncrement”: 4 个重载中没有一个可以转换所有参数类型 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles
       \CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\SlateCore\Public\Rendering\DrawElements.h(724)
       : error C2039: “_InterlockedDecrement”: 不是“FWindowsPlatformAtomics”的成员 [D:\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Source\Runtime\SlateCore\Public\Rendering\DrawElements.h(724)
       : error C2665: “_InterlockedDecrement”: 4 个重载中没有一个可以转换所有参数类型 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles
       \CarlaUE4.vcxproj]
         UnrealBuildTool : error : UBT ERROR: Failed to produce item: D:\carla\Unreal\CarlaUE4\Plugins\Carla\Binaries\W
       in64\UE4-Carla.lib [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.T
       argets(44,5): error MSB3075: 命令“"D:\Program Files (x86)\Epic Games\UE_4.19\Engine\Build\BatchFiles\Build.bat" Ca
       rlaUE4 Win64 Development "D:\carla\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild”已退出，代码为 5。 请验证您是否拥有
       运行此命令的足够权限。 [D:\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]

matthew-rococo, 2018-11-18:
First run 'make clean' before running 'make launch'.  This may deal with your make problem.

nicksunyang, 2018-11-19:
@matthew-rococo thanks for your reply, It can't work.

matthew-rococo, 2018-11-19:
So it’s very strange.
“it can’t work” means the same error message or new error message? I have git  this new version, for the first time , I got the same error as you. But if I ‘make clean’ before ‘make launch’, there will be only one problem. This new problem can be solved by making a little change of the file “LibCarla\source\carla\client\detail\ActorState.h’

nicksunyang, 2018-11-19:
the same error message @matthew-rococo

matthew-rococo, 2018-11-19:
At current time,  you can only compiling this version in Edit mode , run “make launch-editor” will be OK.

nicksunyang, 2018-11-19:
thinks , it's ok

iFuSiiOnzZ, 2018-11-19:
Hi all,

We move to [Build on Windows 0.9.X issue](https://github.com/carla-simulator/carla/issues/647), as the error is also posted there.

Greetings

closed 2018-11-19 by iFuSiiOnzZ