carla-simulator/carla #973: [0.9.1] build error on ubuntu 16.04

hdy117, 2018-11-19:
0.9.1 has many more excellent features. Brilliant work!
When I try to build 0.9.1 on ubuntu 16.04. I got some errors. 
here is the build log. any suggestions?

BuildPythonAPI.sh: Building Python API for Python 2.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing top-level names to source/carla.egg-info/top_level.txt
writing dependency_links to source/carla.egg-info/dependency_links.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
clang-5.0 -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/include/python2.7 -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-2.7/source/libcarla/libcarla.o -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_ENABLE_LIFETIME_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
In file included from source/libcarla/libcarla.cpp:130:
In file included from source/libcarla/SensorData.cpp:9:
In file included from dependencies/include/carla/image/ImageIO.h:9:
In file included from dependencies/include/carla/image/ImageIOConfig.h:50:
In file included from dependencies/include/boost/gil/extension/io/png_io.hpp:35:
In file included from /usr/include/png.h:321:
**/usr/include/pngconf.h:383:12: error: unknown type name '__pngconf'
           __pngconf.h__ in libpng already includes setjmp.h;
           ^
/usr/include/pngconf.h:383:21: error: cannot use dot operator on a type
           __pngconf.h__ in libpng already includes setjmp.h;
                    ^
/usr/include/pngconf.h:384:12: error: unknown type name '__dont__'
           __dont__ include it again.;
           ^
/usr/include/pngconf.h:384:28: error: expected ';' after top level declarator
           __dont__ include it again.;
                           ^**
4 errors generated.
error: command 'clang-5.0' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed

Kevin199Q, 2018-11-23:
sudo pip install libpng libtiff libjpeg

overlapjho, 2018-11-24:
I encountered this error too. @Kevin199Q solution didn't worked for me.

Kevin199Q, 2018-11-26:
@Marwa215  What command did you run when encountered the ImportError?

Marwa215, 2018-11-26:
@Kevin199Q sudo pip install libpng libtiff libjpeg

Marwa215, 2018-11-26:
clang-5.0 -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/include/python2.7 -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-2.7/source/libcarla/libcarla.o -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_ENABLE_LIFETIME_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
In file included from source/libcarla/libcarla.cpp:130:
In file included from source/libcarla/SensorData.cpp:9:
In file included from dependencies/include/carla/image/ImageIO.h:9:
In file included from dependencies/include/carla/image/ImageIOConfig.h:50:
In file included from dependencies/include/boost/gil/extension/io/png_io.hpp:35:
In file included from /usr/include/png.h:321:
/usr/include/pngconf.h:383:12: error: unknown type name '__pngconf'
           __pngconf.h__ in libpng already includes setjmp.h;
           ^
/usr/include/pngconf.h:383:21: error: cannot use dot operator on a type
           __pngconf.h__ in libpng already includes setjmp.h;
                    ^
/usr/include/pngconf.h:384:12: error: unknown type name '__dont__'
           __dont__ include it again.;
           ^
/usr/include/pngconf.h:384:28: error: expected ';' after top level declarator
           __dont__ include it again.;
                           ^
4 errors generated.
error: command 'clang-5.0' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

when I trying sudo pip install libpng libtiff libjpeg I get
marwa@marwa-Z370M-D3H:~$ sudo pip install libpng libtiff libjpeg
[sudo] password for marwa: 
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
marwa@marwa-Z370M-D3H:~$ sudo pip install libpng 
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
marwa@marwa-Z370M-D3H:~$ sudo pip install libpng libtiff libjpeg
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main

Kevin199Q, 2018-11-26:
Sorry for @hdy117 @overlapjho @Marwa215 , the right command should be sudo apt-get install libpng16-dev libtiff5-dev libjpeg-dev . Please refer to  [https://github.com/carla-
https://github.com/carla-simulator/carla/issues/901

Marwa215, 2018-11-26:
Thanks @Kevin199Q Carla begin compiling properly , but encountered another error

creating 'dist/carla-0.9.1-py2.7-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
BuildPythonAPI.sh: Building Python API for Python 3.
Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    from setuptools import setup, Extension
ImportError: No module named 'setuptools'
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

Kevin199Q, 2018-11-26:
@Marwa215 , you don't have setuptools package, please install with `sudo apt-get install python3-setuptools`

Marwa215, 2018-11-26:
yeas I was installing it, 
sudo apt-get install python-setuptools
thanks

Marwa215, 2018-11-26:
So what this error means, sorry if am asking too much

building 'carla.libcarla' extension
clang-5.0 -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/include/python3.5m -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-3.5/source/libcarla/libcarla.o -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_ENABLE_LIFETIME_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.5/source/libcarla/libcarla.o -Ldependencies/lib -L/usr/lib/gcc/x86_64-linux-gnu/7 -o build/lib.linux-x86_64-3.5/carla/libcarla.cpython-35m-x86_64-linux-gnu.so /home/marwa/carla/PythonAPI/dependencies/lib/libcarla_client.a /home/marwa/carla/PythonAPI/dependencies/lib/librpc.a /home/marwa/carla/PythonAPI/dependencies/lib/libboost_filesystem.a /home/marwa/carla/PythonAPI/dependencies/lib/libboost_python35.a -lpng -ljpeg -ltiff /usr/lib/gcc/x86_64-linux-gnu/7/libstdc++.a
x86_64-linux-gnu-gcc: error: /home/marwa/carla/PythonAPI/dependencies/lib/libboost_python35.a: No such file or directory
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

aroongta, 2018-11-26:
I encountered the same error while building this version. Updating the libpng worked for me. us the command:
`sudo apt-get install libpng16-dev`

In case it doesn't work, ensure that you have have the correct version (current latest version is 3.9.0) of cmake:
```
sudo apt-get install build-essential
wget http://www.cmake.org/files/v3.9/cmake-3.9.0.tar.gz
tar xf cmake-3.9.0.tar.gz
cd cmake-3.9.0
./configure
make
```

If still facing an error, you could try ensuring that all the dependencies have been compiled:
Also do a "make clean" and "git pull" before that.
```
`make setup
make LibCarla
make CarlaUE4Editor
make PythonAPI`
```

overlapjho, 2018-11-26:
@aroongta in my case i still get this errpr `dependencies/include/boost/python/detail/wrap_python.hpp:50:11: fatal error: 
      'pyconfig.h' file not found
# include <pyconfig.h>
`

Marwa215, 2018-11-26:
Thanks @aroongta I tried all three solutions but I still getting the same error

aroongta, 2018-11-26:
@Marwa215 
Did you see a **Fatal error: 'version.h' has been modified since the precompiled header.**somewhere.
If yes, that happens from time to time due to Linux updates. Force a rebuild of all the project files with
```
cd Unreal/CarlaUE4/
make CarlaUE4Editor ARGS=-clean
make CarlaUE4Editor
```
It takes a long time but usually fixes the issue for me. Sometimes you might need to reboot.
Try a force rebuild and then compiling all the dependencies followed by a make launch.

hdy117, 2018-11-26:
> Sorry for @hdy117 @overlapjho @Marwa215 , the right command should be sudo apt-get install libpng16-dev libtiff5-dev libjpeg-dev . Please refer to #901.

This works for me. Thanks

Marwa215, 2018-11-26:
@aroongta still suffering the same problem all rebuild done successfully but still can't see this file!!!

"carla/PythonAPI/dependencies/lib/libboost_python35.a: No such file or directory
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1"

aroongta, 2018-11-26:
@Marwa215  I think from the error displayed that the dependencies and requirements for python are missing in your system. Please check if you are using python 3.5.2

Marwa215, 2018-12-03:
Thanks @aroongta I got it, it was simple problem. this file "ibboost_python35.a" was saved to python27 in the destination I removed python2.7 and renamed the file to python35.a then problem is solved.

micmarty, 2018-12-09:
@marcgpuig Thanks, it worked! It should be fixed, because `make package` or `make PythonAPI` is expected to run smoothly without any interaction.

My specs:
`Ubuntu 16.04`
Carla 0.9.1, branch master (commit: `690e343c60471ec94ace55af4e6e91671affa02d`)
Following docs: https://carla.readthedocs.io/en/latest/how_to_build_on_linux/
Using conda: `conda create --name carla-0.9.1-custom-build python=3.5.2 pip`

As of now, the solution is (according to @Marwa215):
```bash
cp PythonAPI/dependencies/lib/libboost_python27.a PythonAPI/dependencies/lib/libboost_python35.a
```

Now it should work!
`make package`

KyunghyunLee, 2018-12-11:
Simply copy **PythonAPI/dependencies/lib/libboost_python27.a** to **PythonAPI/dependencies/lib/libboost_python35.a** and **PythonAPI/dependencies/lib/libboost_python37.a**
solved issue of mine. 

`cp PythonAPI/dependencies/lib/libboost_python27.a PythonAPI/dependencies/lib/libboost_python35.a`
`cp PythonAPI/dependencies/lib/libboost_python27.a PythonAPI/dependencies/lib/libboost_python37.a`

KyunghyunLee, 2018-12-14:
After I build with copied libboost, I got segmentation fault error. 
However, I am not sure it is due to the copied library.

Increasing per-process limit of core file size to infinity.
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
Malloc Size=131076 LargeMemoryPoolOffset=327748 
Malloc Size=131160 LargeMemoryPoolOffset=458936 
Malloc Size=131160 LargeMemoryPoolOffset=590120 
Malloc Size=131160 LargeMemoryPoolOffset=721304 
Malloc Size=321558 LargeMemoryPoolOffset=1042886 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

KyunghyunLee, 2018-12-14:
When you got the same error with me, you should run **Carla*UE4.sh* in **Dist** folder.

rinatdobr, 2018-12-29:
Is is also possible to specify python API version like this:
`make PythonAPI.2`
or 
`make PythonAPI.3`

closed 2019-03-06 by stale[bot]

RunOrVeith, 2019-03-25:
I still have this issue with 0.9.4. 
Python3 boost library is not found. Copying the python2 boost does not work, if you then try to import carla, an error about pyString will be raised

srinikrish22, 2019-04-11:
I am also having the same issue with 0.9.4 version as well as 0.9.5 too. I was earlier using anaconda python and I checked [1054 ](https://github.com/carla-simulator/carla/issues/1054) and used the default from Ubuntu 16.04 but still on 0.9.5 it still gives me the error libboost_python35.a not found

RunOrVeith, 2019-04-12:
It seems to work if you install using system python.
If you use a virtual environment or pyenv or anaconda, it does not generate boost for python3

Derekabc, 2019-04-23:
`cp PythonAPI/carla/dependencies/lib/libboost_python27.a PythonAPI/carla/dependencies/lib/libboost_python35.a`

jingxizc, 2019-05-16:
> It seems to work if you install using system python.
> If you use a virtual environment or pyenv or anaconda, it does not generate boost for python3

You can find libboost_python35.a   in  carla/Build/boost-1.69.0-c7-install/lib/, then copy this file to carla/PythonAPI/carla/dependencies/lib/ ,...Solve this problem

Why this problem happened? I read the script of  "make PythonAPI" ,then found that when execute the  carla/Util/BuildTools/Setup.sh, it failed to copy the libboost_python35.a to PythonAPI/carla/dependencies/lib/ ....

jingxizc, 2019-05-16:
> I am also having the same issue with 0.9.4 version as well as 0.9.5 too. I was earlier using anaconda python and I checked [1054 ](https://github.com/carla-simulator/carla/issues/1054) and used the default from Ubuntu 16.04 but still on 0.9.5 it still gives me the error libboost_python35.a not found
> 
> Edit: my _/usr/bin/env python3 --version_ and _/usr/bin/python3 --version_ both are 3.5.2

You can find libboost_python35.a   in  carla/Build/boost-1.69.0-c7-install/lib/, then copy this file to carla/PythonAPI/carla/dependencies/lib/ ,...Solve this problem

And in python3.6,  import carla , no error happened , u can have a try

KyunghyunLee, 2019-05-17:
If you are using anaconda and automatically activate it in `bashrc`, etc., remove it and try again.

deepakreddy63, 2019-05-30:
#199
This may help

h042910276, 2019-12-01:
If you can't find libboost_python3x.a in carla/Build/boost-1.69.0-c7-install/lib/
or
libboost_python3x.a is not the version that you want, then:
1. Delete the "carla/Build/boost-1.69.0-c7-install" folder
2. switch the python version which you want to use
3. `make setup`
You will see libboost_python3x.a is in carla/Build/boost-1.69.0-c7-install/lib/

gsujansai, 2020-06-06:
If anyone is facing this problem in 0.9.9, the solution of copying libboost_python27.a suggested by @KyunghyunLee and @Derekabc  solves the problem. Thanks guys.